fieldOfView/Cura-OctoPrintPlugin #167: Extrude button in monitor tab

gkrystev, 2020-04-17:
In the "Monitor" tab in Cura, it will be useful to have "extrude" and "retract" buttons near the "Jog position" buttons. At least, in my opinion, it will be useful when changing filament. So far I'm doing this operation by sending gcode or go to the OctoPrint web interface.

fieldOfView, 2020-04-17:
I agree with the feature request. What has stopped me sofar from implementing this request is that I can't think of a good design that integrates this with the existing extruder temperature boxes. But perhaps I should let go of my pride and go for the functional design, similar to what OctoPrint itself has.

thelastWallE, 2020-04-22:
Just to hook on this enhancement.
I was just thinking that a button for turn on/off the printer would be nice in the monitoring tab.
You do this already with the tplink_octoprint_plugin before gcode upload to octoprint.

fieldOfView, 2020-04-22:
Don’t “hook” multiple requests together, please

thelastWallE, 2020-04-22:
Sry, was just thinking that it's fairly similar because of the monitoring tab.
I will open a new "issue".
Good day to you and thanks for all your work on cura!

fieldOfView, 2020-04-22:
@thelastWallE, see https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/72