fieldOfView/Cura-OctoPrintPlugin #170: Breaking Change with UFP Plugin

jneilliii, 2020-04-28:
@fieldOfView, I was testing the development snapshot I saw in another post on my latest development version of UFP plugin to test auto-start printing and there's a breaking change on my side. The file name will no longer include the .ufp in the gcode filename and will be just <filename>.gcode the same as if you just uploaded a gcode variant. I've also removed the forcing of analysis queue to hopefully help with the PTG issue. You can get my latest version using the link below.

 https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage/archive/0.1.7.zip

fieldOfView, 2020-04-28:
> I've also removed the forcing of analysis queue to hopefully help with the PTG issue.

So now analysis doesn't take place at all for ufp uploads?

jneilliii, 2020-04-28:
It does. I just converted the plugin to use OctoPrint's internal add_file function instead of extracting the gcode directly to uploads and forcing the analysis.

fieldOfView, 2020-04-28:
Is there any way I can recognise this version of the plugin, or query if the plugin will create a .ufp.gcode or a .gcode file?

jneilliii, 2020-04-28:
I'm not 100% sure @fieldOfView, I assume there is some API call that will tell you the version, maybe even the same call that you are using to know if the plugin is installed? I just don't know.

jneilliii, 2020-04-28:
I am seeing a bug however with the current dev version that I just noticed that may be related to this move to add_file. Not sure if this is windows specific or not but it seems like it might be while setting additional metadata while analysis is ongoing. Was about to post to community forum for some guidance.

```
2020-04-28 10:26:43,901 - octoprint.filemanager.storage - ERROR - Error while writing .metadata.json to C:\Users\jneil\AppData\Roaming\OctoPrint\uploads
Traceback (most recent call last):
  File "c:\octoprint\venv\lib\site-packages\octoprint\filemanager\storage.py", line 1583, in _save_metadata
    f.write(to_bytes(json.dumps(metadata, indent=4, separators=(",", ": "))))
  File "c:\python27\Lib\contextlib.py", line 24, in __exit__
    self.gen.next()
  File "c:\octoprint\venv\lib\site-packages\octoprint\util\__init__.py", line 1105, in atomic_write
    shutil.move(temp_config.name, filename)
  File "c:\python27\Lib\shutil.py", line 317, in move
    os.unlink(src)
WindowsError: [Error 32] The process cannot access the file because it is being used by another process: u'c:\\users\\jneil\\appdata\\local\\temp\\tmpsc10ri'
```

fieldOfView, 2020-04-28:
>  maybe even the same call that you are using to know if the plugin is installed?

To detect the plugin, I check if there are settings for it in `/api/settings`. Back when your plugin did not have settings yet, I even made a PR to add a dummy setting (installed = true) so there would be something to detect. Perhaps you could add a readonly setting that shows the version?

jneilliii, 2020-04-28:
Of course, no problem. Will add that in my next round of updates to this development version.

jneilliii, 2020-04-28:
I just pushed an update to that same branch with a new setting `installed_version` for your use. It will always contain the latest version of the plugin.

jneilliii, 2020-04-29:
I can confirm this latest commit is working as expected. Sorry for the name change.

fieldOfView, 2020-04-29:
Thanks for testing the fix.

fieldOfView, 2020-06-04:
Version 3.5.13 is now available in the Marketplace.

closed 2020-06-04 by fieldOfView

jneilliii, 2020-06-04:
Thanks @fieldOfView