fieldOfView/Cura-OctoPrintPlugin #171: Preheat bed uses wrong temperature

dxgldotorg, 2020-05-22:
I have my Cura settings with a higher initial temperature for the bed to help the first layer stick better but drop my temperature after the first layer.  The preheat button, instead of using the "Build Plate Temperature Initial Layer" setting, uses the "Build Plate Temperature" setting to preheat.

fieldOfView, 2020-05-22:
This would need to be fixed in Cura, not in this plugin. 

It would not be a hard thing to fix, but I am not 100% sure if it should be done. It would cause a lot of confusion for printers where the initial build plate temperature is always 5 degrees higher than the normal build plate temperature. In those cases, users will wonder why Cura uses a different temperature for preheating than what they have entered.

As a workaround, you can always change the value that will be used for preheating.

dxgldotorg, 2020-05-22:
Thanks; I was thinking about reporting it in Cura but figured it was the addon's issue.

closed 2020-05-22 by dxgldotorg