fieldOfView/Cura-OctoPrintPlugin #173: Crash in OctoPrintPowerPlugins: all_config_set not defined

Ghostkeeper, 2020-06-05:
We're getting a new crash since yesterday in this plug-in: https://sentry.io/share/issue/264817437d7f4a75a83d9eadb4213cfe/

It looks like the `all_config_set` variable is not defined if `additional_data` is empty.

Ghostkeeper, 2020-06-05:
10 minutes! Lightning fast!

fieldOfView, 2020-06-05:
Already fixed by @martinh2011

fieldOfView, 2020-06-05:
I like keeping the issue open until a fix is available in the Marketplace.

I can't see on the sentry link how often the crash is occurring, can you?

Ghostkeeper, 2020-06-05:
Yeah, I can see it. We've seen 35 crashes so far. That's quite a lot for 1 day.

fieldOfView, 2020-06-08:
I submitted a fixed version to the Marketplace last friday; I hope it will be available there soon. Until it is, you can download the fixed version manually. Drop the downloaded [curapackage](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/OctoPrintPlugin/6NH1d-JS57OVwXNOEKkviyExzatNoGko_Z1LKubUY7A5/OctoPrintPlugin-v7.2.0-2020-06-05T10%3A22%3A50Z.curapackage?Expires=1906957345&GoogleAccessId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=l8Y7ZJ4WPd3uSxuLXScwMgg0GhJogllGtizkcy0zqFOOqEeZ0lqU9GVIj%2FdhmZsFMf1bfh%2FnT898MByjpGoLnKRMPzAl6CNsAQXUzbNwwXoydJSuPkY25l2t0oV0a5vCg7QQiiaD5i0feJjm7SC%2FEpEguyorYkYV6OBoZGBCAkF4RbgALJgxNy3Fc%2BMz35ezIvYC8jlifJKQXRsEEZ6YoT9DFLgk2fdqZZy80AWSlCLzpJrrlvGIQVNB2gItaO%2Fqdf5UfSyrYt%2B1VuXUcPb6lVrWamPwWfEz6yTGAlepvXL0IMcKUF6d9rK2v7UkVycn4Xhya34AUEyOomu%2FRkz3Mg%3D%3D) onto the Cura build plate as if you were opening a 3d model.

mahtDFR, 2020-06-08:
3.5.14 is through the marketplace now

closed 2020-06-08 by fieldOfView