fieldOfView/Cura-OctoPrintPlugin #175: Error file not found on sd card

devildant, 2020-06-10:
Hello,

I note a small bug with the last version, an error is raised systematically ("Error file not found..."), when the option "store gcode on the printer sd card" is activated.

cura 4.6.1
plugin version : 3.5.14
[image]

fieldOfView, 2020-06-10:
Thanks for the report. I'll try to find out what is going wrong.
 
Could you try if the error also happens if you have the "Automatically start print job after uploading" option selected?

devildant, 2020-06-10:
hello, I tell you after my print in 7 hours :)

devildant, 2020-06-10:
I was able to test, same error with the ""Automatically start print job after uploading" option activated (the print starts well), so it's also a false positive

closed 2020-07-16 by fieldOfView