fieldOfView/Cura-OctoPrintPlugin #178: timelapse option

Glutius, 2020-06-12:
Whenever I start a print from the octoprint's UI, timelapse recording automatically starts, but whenever I start a print from Cura, it doesn't start a timelapse.

Don't yet not if this is implimented or not, or if there's a setting for this is octoprint's settings, but couldn't find anything. Love literally everything about this plugin but this is the one thing that's missing imo.

fieldOfView, 2020-06-12:
I'll see what is going on, and if there is anything I can do about it.

Glutius, 2020-06-13:
Started using octolapse plugin on octoprint, and whenever I start now a print from Cura that starts as well with flawless results, so sort of, it's solved.

fieldOfView, 2020-06-14:
I hope you don't mind if I keep this issue open for a while longer, until I have time to see if there is anything I can do to fix the problem for the standard timelapse functionality.

fieldOfView, 2020-06-15:
I have looked in to it, and came to the conclusion that it should work, and if it does not work then it should be fixed in OctoPrint. 

OctoPrint starts the timelapse when a print is started here:
https://github.com/OctoPrint/OctoPrint/blob/6232a04e52fbf09a8b912590718764fccc3598ba/src/octoprint/timelapse.py#L464
That method is triggered by the `Events.PRINT_STARTED` event. If the timelapse does not start when the gcode file is uploaded through the API, then that means that that event does not fire (?!).

closed 2020-06-15 by fieldOfView