fieldOfView/Cura-OctoPrintPlugin #179: Camera error "Unusable stream url"

alexeyzel, 2020-07-01:
I use Cura 4.6.1 and OctoPrint 1.4.0.
I connected camera from my local MotionEye server using Streaming URL. The camera shows in OctoPrint but is not showing in Cura.
My plugin version is 3.5.14.
In cura.log i have following error messages:

```
2020-07-01 21:52:15,143 - DEBUG - [MainThread] cura.CuraApplication.run [799]: Booting Cura took 11.432587385177612 seconds
2020-07-01 21:52:15,147 - INFO - [MainThread] UM.OutputDevice.OutputDeviceManager.addOutputDevice [171]: Output Device "Flsun Q5"._octoprint._tcp.local already added
2020-07-01 21:52:15,149 - WARNING - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice.parseSettingsData [1182]: Unusable stream url received:  http://192.168.1.21:8083
2020-07-01 21:52:15,149 - DEBUG - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice.parseSettingsData [1185]: Set OctoPrint camera url to 
2020-07-01 21:52:15,167 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._processRequest [346]: Request [0f940e225ca24a4a8c86858838e61a94] started
```

fieldOfView, 2020-07-02:
Could you post the output of `http://[your octoprint ip address]/api/settings?apikey=[your api key]` ? Specifically the `webcam` section.

fieldOfView, 2020-07-02:
It looks like there is a space in front of the url you configured in OctoPrint. Could you check if that is the case?

alexeyzel, 2020-07-02:
`,"webcam":{"bitrate":"10000k","ffmpegPath":"/usr/bin/ffmpeg","ffmpegThreads":1,"ffmpegVideoCodec":"mpeg2video","flipH":false,"flipV":false,"rotate90":false,"snapshotSslValidation":true,"snapshotTimeout":5,"snapshotUrl":"http://127.0.0.1:8080/?action=snapshot","streamRatio":"16:9","streamTimeout":5,"streamUrl":" http://192.168.1.21:8083","timelapseEnabled":false,"watermark":false,"webcamEnabled":true}}`

Yes, there was a space in URL, OctoPrint showed no error and worked fine. After removing space in URL - it works in OctoPrint and in Cura Plugin.

fieldOfView, 2020-07-02:
I will make the next version of the plugin robust against this configuration error.

closed 2020-07-16 by fieldOfView