fieldOfView/Cura-OctoPrintPlugin #180: Feature request: Flip camera horizontally/vertically

JamieS01, 2020-07-04:


fieldOfView, 2020-07-04:
The OctoPrint Connection plugin applies the same flips and rotation that is configured in OctoPrint. Is your camera showing in the right orientation in the OctoPrint webinterface?

JamieS01, 2020-07-05:
> The OctoPrint Connection plugin applies the same flips and rotation that is configured in OctoPrint. Is your camera showing in the right orientation in the OctoPrint webinterface?

It shows correctly in the octoprint interface, but not in the plugin.

fieldOfView, 2020-07-05:
Please visit `http://[address of your octoprint instance]/api/settings?apikey=[your api key]` and post the `webcam` section of the file you get, which looks like this:

[image]

JamieS01, 2020-07-06:
{"api":{"allowCrossOrigin":false,"key":"3E2674A51FB34DF2B01226BB3C0C9FEE"},"appearance":{"closeModalsWithClick":true,"color":"orange","colorIcon":true,"colorTransparent":false,"defaultLanguage":"_default","fuzzyTimes":true,"name":"Ender 3","showFahrenheitAlso":false},"feature":{"autoUppercaseBlacklist":["M117","M118"],"g90InfluencesExtruder":false,"gcodeViewer":true,"keyboardControl":true,"mobileSizeThreshold":2097152,"modelSizeDetection":true,"pollWatched":false,"printCancelConfirmation":true,"printStartConfirmation":false,"sdSupport":true,"sizeThreshold":20971520,"temperatureGraph":true},"folder":{"logs":"/home/pi/.octoprint/logs","timelapse":"/home/pi/.octoprint/timelapse","timelapseTmp":"/home/pi/.octoprint/timelapse/tmp","uploads":"/home/pi/.octoprint/uploads","watched":"/home/pi/.octoprint/watched"},"gcodeAnalysis":{"runAt":"idle"},"plugins":{"BLTouch":{"cmdProbeBed":"G29","cmdProbeDown":"M280 P0 S10","cmdProbeUp":"M280 P0 S90","cmdReleaseAlarm":"M280 P0 S160","cmdSaveSettings":"M500","cmdSelfTest":"M280 P0 S120","confirmation":true},"HeaterTimeout":{"enabled":true,"interval":15,"notifications":true,"timeout":1800},"PrintTimeGenius":{"allowAnalysisWhileHeating":true,"allowAnalysisWhilePrinting":false,"analyzers":[{"command":"\"/home/pi/oprint/bin/python2\" \"/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py\" \"{gcode}\"","description":"All gcode analyzers (usually not as good as marlin-calc)","enabled":false},{"command":"\"/home/pi/oprint/bin/python2\" \"/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py\" marlin-calc \"{gcode}\" \"{mcodes}\"","description":"Marlin firmware simulation (replaces Octoprint built-in, faster and more accurate)","enabled":true},{"command":"\"/home/pi/oprint/bin/python2\" \"/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py\" \"{gcode}\" --parsers slic3r_pe_print_time slic3r_pe_print_time_remaining","description":"Use Slic3r PE M73 time remaining","enabled":false}],"enableOctoPrintAnalyzer":false,"exactDurations":true},"TerminalCommands":{"controls":[{"commands":"M503","name":"M503 (Retrieve E-Steps etc.)"},{"commands":"G91","name":"Relative Extrusion"},{"commands":"M500","name":"Save to EEPROM"},{"commands":"G1 E100 F100","name":"Extrude 100mm"}]},"action_command_prompt":{"command":"M876","enable":"detected","enable_emergency_sending":true,"enable_signal_support":true},"announcements":{"channel_order":["_important","_releases","_blog","_plugins","_octopi"],"channels":{"_blog":{"description":"Development news, community spotlights, OctoPrint On Air episodes and more from the official OctoBlog.","name":"On the OctoBlog","priority":2,"read_until":1593102540,"type":"rss","url":"https://octoprint.org/feeds/octoblog.xml"},"_important":{"description":"Important announcements about OctoPrint.","name":"Important Announcements","priority":1,"read_until":1521111600,"type":"rss","url":"https://octoprint.org/feeds/important.xml"},"_octopi":{"description":"News around OctoPi, the Raspberry Pi image including OctoPrint.","name":"OctoPi News","priority":2,"read_until":1573722900,"type":"rss","url":"https://octoprint.org/feeds/octopi.xml"},"_plugins":{"description":"Announcements of new plugins released on the official Plugin Repository.","name":"New Plugins in the Repository","priority":2,"read_until":1593838800,"type":"rss","url":"https://plugins.octoprint.org/feed.xml"},"_releases":{"description":"Announcements of new releases and release candidates of OctoPrint.","name":"Release Announcements","priority":2,"read_until":1583322300,"type":"rss","url":"https://octoprint.org/feeds/releases.xml"}},"display_limit":3,"enabled_channels":["_octopi","_plugins","_releases","_blog"],"forced_channels":["_important"],"summary_limit":300,"ttl":360},"bedlevelvisualizer":{"command":"G28\nM155 S30\n@BEDLEVELVISUALIZER\nG29 T\nM155 S3\nM500","commands":[],"debug_logging":false,"descending_x":false,"descending_y":false,"flipX":false,"flipY":false,"ignore_correction_matrix":false,"imperial":false,"mesh_timestamp":"04/07/2020, 13:34:08","mesh_unit":1,"reverse":false,"rotation":0,"save_mesh":true,"screw_hub":0.5,"show_labels":true,"show_webcam":false,"showdegree":false,"stored_mesh":[["-0.145","-0.060","-0.000","+0.080","+0.122"],["-0.215","-0.120","-0.040","+0.055","+0.110"],["-0.205","-0.108","-0.005","+0.105","+0.177"],["-0.130","-0.028","+0.040","+0.132","+0.172"],["-0.020","+0.022","+0.082","+0.145","+0.182"]],"stored_mesh_x":[0,55,110,165,220],"stored_mesh_y":[0,55,110,165,220],"stored_mesh_z_height":250,"stripFirst":false,"timeout":1800,"use_center_origin":false,"use_relative_offsets":false},"costestimation":{"costOfElectricity":0.25,"costOfFilament":"10","currency":"\u00a3","currencyFormat":"%v %s","densityOfFilament":1.32,"diameterOfFilament":1.75,"lifespanOfPrinter":0,"maintenanceCosts":0,"powerConsumption":0.2,"priceOfPrinter":0,"requiresLogin":false,"useFilamentManager":true,"weightOfFilament":1000},"discordremote":{"allow_scripts":false,"allowed_gcode":"","baseurl":"http://octopi.local","bottoken":"NzE3MDYzNzI0ODIwODU2OTE1.XtU4KQ.hkUVjdXGssTNq65XU2m17Fp4HxU","channelid":"717063537410965564","events":{"printer_state_error":{"enabled":true,"message":"\u26a0\ufe0f Your printer is in an erroneous state.","name":"Printer state : error","with_snapshot":false},"printer_state_operational":{"enabled":true,"message":"\u2705 Your printer  [... cut ...] : SD printing byte)|(Recv: Not SD printing)"},{"name":"Suppress wait responses","regex":"Recv: wait"}],"webcam":{"bitrate":"10000k","ffmpegPath":"/usr/bin/ffmpeg","ffmpegThreads":1,"ffmpegVideoCodec":"mpeg2video","flipH":true,"flipV":true,"rotate90":false,"snapshotSslValidation":true,"snapshotTimeout":10,"snapshotUrl":"http://localhost:8080/?action=snapshot","streamRatio":"4:3","streamTimeout":5,"streamUrl":"/webcam/?action=stream","timelapseEnabled":true,"watermark":false,"webcamEnabled":true}}

fieldOfView, 2020-07-06:
I can't reproduce this with the current version of Cura. I refactored the camera handling to clean up the code and support multiple cameras a little while ago. Could you test this [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/OctoPrintPlugin/JplGzXeqTmvItgk-5_Ee7zb_iQLS7WR6dowqxU91AGGZ/OctoPrintPlugin-v7.0.0-2020-07-03T10%3A06%3A19Z.curapackage?Expires=1909406505&GoogleAccessId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=UXcQDNfZ8r4RPqn0As5UAyGvFSjPXKQjtk9AX2PPGJavNkNmu8ls3q6ccamNe4c01yDjHwW3%2FUXyzlVh3hKVgRXmZ6hfhVTA4KW0JXdGmmA3GPN4V0MpkcIiR5EjmyEW6y8b5uJJL0vbr4tISsNiTUUzPVBzOIP4ykZRWctF2R8jKHG1ifeNesfQ4syl2l2T%2FfTtIG%2BFvIcOaOBlfQUa2Mqaq27cXioFhveQypWehiUfGONYh2sxEcorV%2BpLR8mnaDk6xiMXJF%2FdzgCH1hBc6c%2F6JvhaxbFhbG5pVLXPVPV9oOQkZ2Lu6Lw2ErNflfxt%2B0VzYaGIzbRVq5vCVF3ZHQ%3D%3D)? Download the file, and drop it onto the buildplate in Cura as if you were opening a 3d model through drag and drop.

JamieS01, 2020-07-06:
Seems to be working with that snapshot! Is there any other information that I can provide which may help?
[image]

fieldOfView, 2020-07-07:
> Seems to be working with that snapshot!

[image]

Well, it seems I am going in the right direction then. I will add a "fixed" label to this issue, but will keep the issue open until a version containing this fix is available in the Marketplace.

closed 2020-07-16 by fieldOfView