fieldOfView/Cura-OctoPrintPlugin #181: Crash when webcam's streamUrl is None

Ghostkeeper, 2020-07-17:
Hi,

We're seeing this crash report crop up a couple of times, since today: https://sentry.io/share/issue/f56f8c9106d04fc2959230c7db6d363c/

It seems that the webcam data has a `streamUrl` entry that is set to `None`. The "strip" function then doesn't work and crashes Cura. Maybe this needs to have a check for `None` which sets it to an empty string then?
I don't know where the `None` comes from though; perhaps that is where the problem lies.

Ghostkeeper, 2020-07-17:
Since that sharing link is not as expansive as what the developers can see, here is a screenshot of some more information:
[image]

And we also get a traceback:
[image]

This was seen 16 times so far.

fieldOfView, 2020-07-17:
Thanks for the report. I can see how this happens, and it is easy to fix. Very unfortunate that it slipped through testing. Do you think it is worth pulling the plugin from the marketplace over?

fieldOfView, 2020-07-17:
I have submitted a fixed version to the Marketplace.

commit e2c128bc (2020-07-17): Fix crash when no webcam is configured

Fixes #181
closed 2020-07-30 by fieldOfView