fieldOfView/Cura-OctoPrintPlugin #182: Compatibility with GCode Filename Format

StumbleNOLA, 2020-07-21:
Would it be possible to have OctoPrint Connection use the file format generated by "GFF"? I need GFF to keep my files organized and properly named, but the two don't play together, so I end up not using OPC for anything but one time print files I plan on deleting.

fieldOfView, 2020-07-22:
Because of the design of the plugin, unfortunately no. You could use the [Custom Printjob Naming](https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/CustomJobPrefix) plugin instead.

Perhaps a future version of the GCode Filename Format plugin will change this:
https://github.com/rgomezjnr/GcodeFilenameFormat/issues/2

closed 2020-07-22 by fieldOfView

rgomezjnr, 2020-07-29:
Hey guys just saw this and wanted to add. I added compatibility for OcotoPrint Connection in GFF in version [2.0.0-rc2](https://github.com/rgomezjnr/GcodeFilenameFormat/releases/tag/2.0.0-rc2). Because of the significant changes and unfinished testing I haven't integrated this into a release, but other users have reported that it works for them. This version removes the separate Save Gcode button, and actively updates the job name with the specified format, which will then be used by OctoPrint Connection. @StumbleNOLA I can help you install this version if you would like.

StumbleNOLA, 2020-07-29:
@rgomezjnr That would be great, I am happy to beta test it. I hate having to bounce between the two plugins.

rgomezjnr, 2020-07-29:
1. Download [2.0.0-rc2](https://github.com/rgomezjnr/GcodeFilenameFormat/releases/tag/2.0.0-rc2).
2. Extract 2.0.0-rc2
3. Move GcodeFilenameFormat folder to Cura plugins folder
- Windows: C:\Users\\%USERNAME%\AppData\Roaming\cura\4.4\plugins
- Mac: $User/Library/Application\ Support/Cura/4.4/plugins
- Linux: $HOME/.local/share/cura/4.4/plugins
3. Launch Cura
4. Set filename format 
5. Slice
6. Select Print with OctoPrint button
7. Verify the filename format is used In OctoPrint Connection

StumbleNOLA, 2020-08-05:
Sorry I didn’t have time to play with this previously. 

I am having an issue getting the plugin installed. I assume it is an issue on my end, but I am not sure what I am doing wrong.

I opened Cura, and removed the old GcodeFilenameFormat
Then closed Cura
Unzipped the files
Copied the files into the Cura/plugins folder (Including GcodeFilenameFormat-2.0.0-rc2

But when I reload Cura it doesn’t appear to be there. If I go to Extensions there is nothing there. And if I click on Marketplace - Installed all I have is Octoprint, and a greyed out Gcode… 1.4.1. If I click on the box it checks, and tells me to reload Cura. But on reload nothing happens.

I am pretty sure this is just me not knowing Cura well, but can you let me know what I am doing wrong?

Greg

fieldOfView, 2020-08-06:
Is it perhaps a good idea to move this discussion to https://github.com/rgomezjnr/GcodeFilenameFormat/issues/2 ?

rgomezjnr, 2020-08-07:
Yes @fieldOfView, I agree.

@StumbleNOLA please post further questions on this in [rgomezjnr/GcodeFilenameFormat#2 ](https://github.com/rgomezjnr/GcodeFilenameFormat/issues/2).

I suggest first uninstalling the plugin using the Cura Marketplace:

[image]

Once that is done, close Cura and verify there is no GcodeFilenameFormat folder in your Cura plugins folder. If there is then delete it.

Then extract and copy the GcodeFilenameFormat folder to the Cura configuration folder. Note there shouldn't be any "-2.0.0-rc2" in the folder name.

You should have the following paths:

```
cura\4.4\plugins\GcodeFilenameFormat\images
cura\4.4\plugins\GcodeFilenameFormat\test
cura\4.4\plugins\GcodeFilenameFormat\__init__.py
cura\4.4\plugins\GcodeFilenameFormat\Format.qml
cura\4.4\plugins\GcodeFilenameFormat\GcodeFilenameFormat.py
cura\4.4\plugins\GcodeFilenameFormat\Help.qml
cura\4.4\plugins\GcodeFilenameFormat\LICENSE
cura\4.4\plugins\GcodeFilenameFormat\plugin.json
cura\4.4\plugins\GcodeFilenameFormat\README.md
```

If you are sure the plugins folder is correct, then launch Cura. If it's still not loading the plugin, check the Cura log file in:

`cura\4.6\cura.log`

If the plugin is loading properly it should report the following:

2020-08-06 21:12:47,282 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SimpleView
2020-08-06 21:12:47,312 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin X3DReader
2020-08-06 21:12:47,352 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin XmlMaterialProfile
2020-08-06 21:12:47,375 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin XRayView
**2020-08-06 21:12:47,456 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GcodeFilenameFormat**

If there's an error after it attempts to load GcodeFilenameFormat please paste it in [rgomezjnr/GcodeFilenameFormat#2 ](https://github.com/rgomezjnr/GcodeFilenameFormat/issues/2).