fieldOfView/Cura-OctoPrintPlugin #186: Folder support

ollien, 2020-08-09:
One of the features I like about the PrusaSlicer octoprint feature is that I can send all of my uploads to an "uploads" folder to at least keep my uploaded prints from my sorted ones. It would be super cool if I could set a default upload location for this plugin. 

Happy to make a PR for it myself, just don't wanna make it if you're totally opposed to the concept :)

fieldOfView, 2020-08-09:
If you install the "Custom Printjob Naming" plugin, you can configure a path via that plugin:
https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/CustomJobPrefix

The path configured via that plugin can even have several replacement patterns in it, if you wish.

fieldOfView, 2020-08-09:
Duplicate of https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/28

closed 2020-08-09 by fieldOfView

ollien, 2020-08-09:
Oh neat. Sorry, can't believe I didn't see that issue. Thanks :)