fieldOfView/Cura-OctoPrintPlugin #187: Error "Printer is already printing, cannot select a new file"

fmjensen, 2020-08-11:
When I click print with octoprint I get this error

[image]

The printer starts the print none the less so this is not a show stopper.

This started after i updated OctoPrint to Version 1.4.2

Cura.log shows this:
```
2020-08-11 09:29:19,486 - DEBUG - [MainThread] UM.Controller.setActiveStage [180]: Setting active stage to MonitorStage
2020-08-11 09:29:19,494 - DEBUG - [MainThread] UM.Controller.setActiveView [108]: Setting active view to SolidView
2020-08-11 09:29:19,502 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.6/plugins/PreviewStage/PreviewMenu.qml:25: TypeError: Cannot read property of null
2020-08-11 09:29:19,504 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.6/plugins/PreviewStage/PreviewMenu.qml:26: TypeError: Cannot read property of null
2020-08-11 09:29:19,521 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.6/resources/qml/ActionButton.qml:101: ReferenceError: contentWidth is not defined
2020-08-11 09:29:19,523 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.6/resources/qml/ActionButton.qml:101: ReferenceError: contentWidth is not defined
2020-08-11 09:29:19,584 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._processRequest [346]: Request [eb0cc4270b63494e93859cfbb7787bc2] started
2020-08-11 09:29:19,586 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [317]: No more requests to process, stop
2020-08-11 09:29:19,611 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.6\resources\shaders\xray.shader]...
2020-08-11 09:29:20,438 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestFinished [408]: Request [eb0cc4270b63494e93859cfbb7787bc2] finished.
2020-08-11 09:29:20,438 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [317]: No more requests to process, stop
2020-08-11 09:29:20,438 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [275]: SliceInfo sent successfully
2020-08-11 09:29:20,546 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.6/resources/qml/RoundedRectangle.qml:68: TypeError: Cannot read property of null
2020-08-11 09:29:20,778 - DEBUG - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onUploadFinished [1102]: Resource created on OctoPrint instance: http://10.0.1.10/api/files/local/CE5P_halls_downpipe_with_mesh.gcode
2020-08-11 09:29:20,797 - WARNING - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [837]: Encountered unexpected printjob state: Starting
2020-08-11 09:29:20,915 - ERROR - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [1045]: OctoPrintOutputDevice got an error while accessing http://10.0.1.10:80/api/files/local/CE5P_halls_downpipe_with_mesh.gcode
2020-08-11 09:29:20,916 - ERROR - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [1046]: Printer is already printing, cannot select a new file
2020-08-11 09:29:29,695 - DEBUG - [MainThread] cura.AutoSave._onTimeout [60]: Autosaving preferences, instances and profiles
```

Let me know if you need more information.

fieldOfView, 2020-08-11:
Thanks for the report. I'll try to see what is going wrong.

gdz451, 2020-08-24:
This worked for me.

tigert, 2020-08-24:
The same dialog popped up for me also, even though it appeared to print though, as if it was trying twice.

Your snapshot fixed it for me also, thanks for that, and thanks for working on this plugin, much appreciated.

fieldOfView, 2020-08-24:
Great, I'll see about making a new release including this fix soon.

ciscomike1, 2020-09-04:
After updating my Cura to 4.7.0 had the same issue.
Solved with this development snapshot as well.

mrevilspirit, 2020-09-10:
same problem cura 4.7.1

AdmiralMichael, 2020-09-11:
Just to add I have the same issue. 

Cura 4.7.1 for MacOS
OctoPrint 1.4.2 Python 3.7.3 OctoPi 0.17.0

Nut I just want to add that I **do not** have the following enabled:
"Automatically start print jobs after uploading"
or
"Select Print Job after uploading"

It's just strictly set to upload the file and I still get that error.

fieldOfView, 2020-09-12:
@mrevilspirit and @AdmiralMichael did you try the development snapshot linked to above?

I have submitted a fixed version to the Marketplace, but it is awaiting approval.

AdmiralMichael, 2020-09-14:
Not yet. I just wanted to point out I was getting g the error despite the fact I don’t have “Automatically start print jobs after uploading" or
"Select Print Job after uploading" selected.

fieldOfView, 2020-09-15:
It would be helpful if you did. Otherwise I only get feedback if my fixes work after making another release.

AdmiralMichael, 2020-09-15:
Ok so I have tired the development snapshot and I still get the "Printer is already running.." message. I even removed the plugin and installed the snapshot. Cura reports the OctoPrint plugin as v3.5.17 if that means anything.

aledrus, 2020-09-16:
Hi. The development snapshot got rid of the "printer is already running" message for me. The 4.7.1 Cura update still gave me errors before this.

fieldOfView, 2020-09-16:
Fix is available in version 3.5.17 in the Marketplace, but I am not sure if it also fixes things for @AdmiralMichael. His might be a slightly different issue.

@AdmiralMichael could you try with the version that is currently in the Marketplace? If that still does not fix it for you, I will need your cura.log. See help->show configuration files.

fieldOfView, 2020-09-21:
@AdmiralMichael I need your Cura.log to fix this issue for you.

AdmiralMichael, 2020-09-21:
I dont know what happened but I no longer get that error. Everything seems to work fine now.

fieldOfView, 2020-09-21:
Great, thanks for confirming that here. Don't be shy when it happens again! I'll close the issue as "fixed" for now.

closed 2020-09-24 by fieldOfView

AdmiralMichael, 2020-10-03:
So I just tried to upload a new project and the error is back. How can I send you my cura.log? Is it safe to upload it here?

mikekscholz, 2020-10-13:
I'm having a similar issue , getting the same pop-up dialog, but I don't have any of the check boxes enabled on my octoprint connection. By my logs it appears to be trying to start the print anyway, and failing with the already printing message when the printer is in fact offline. 

```
2020-10-13 15:45:20,408 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [710]: Slicing took 2.2331759929656982 seconds
2020-10-13 15:45:20,408 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [711]: Number of models per buildplate: {0: 1}
2020-10-13 15:45:20,409 - DEBUG - [Thread-10] CuraEngineBackend.ProcessSlicedLayersJob.run [74]: Processing new layer for build plate 0...
2020-10-13 15:45:20,635 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [725]: See if there is more to slice...
2020-10-13 15:45:20,641 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [196]: Socket state changed to Closing
2020-10-13 15:45:20,647 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [220]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
2020-10-13 15:45:20,650 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [198]: Socket state changed to Closed
2020-10-13 15:45:20,658 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [898]: Backend quit with return code 0. Resetting process and socket.
2020-10-13 15:45:20,939 - DEBUG - [Thread-10] CuraEngineBackend.ProcessSlicedLayersJob.run [254]: Processing layers took 0.5297281742095947 seconds
2020-10-13 15:45:20,945 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onProcessLayersFinished [929]: See if there is more to slice(2)...
2020-10-13 15:45:26,341 - DEBUG - [MainThread] SliceInfoPlugin.SliceInfo._onWriteStarted [114]: 'info/send_slice_info' is turned off.
2020-10-13 15:45:26,426 - DEBUG - [MainThread] UFPWriter.UFPWriter._createSnapshot [45]: Creating thumbnail image...
2020-10-13 15:45:26,430 - DEBUG - [MainThread] UM.Scene.Camera._preferencesChanged [255]: Changing perspective mode to perspective.
2020-10-13 15:45:26,434 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\overhang.shader]...
2020-10-13 15:45:26,439 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.7\resources\shaders\striped.shader]...
2020-10-13 15:45:26,936 - DEBUG - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice.requestWrite [465]: Tried starting a print, but current state is offline
2020-10-13 15:45:27,774 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.7/resources/qml/RoundedRectangle.qml:68: TypeError: Cannot read property of null
2020-10-13 15:45:37,686 - DEBUG - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onUploadFinished [1109]: Resource created on OctoPrint instance: http://172.20.10.9/api/files/local/screen_front_thinned.ufp
2020-10-13 15:45:39,131 - ERROR - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [1052]: OctoPrintOutputDevice got an error while accessing http://172.20.10.9:80/api/files/local/screen_front_thinned.gcode
2020-10-13 15:45:39,131 - ERROR - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [1053]: Printer is already printing, cannot select a new file
2020-10-13 15:45:45,598 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.7/resources/qml/RoundedRectangle.qml:68: TypeError: Cannot read property of null
```

pferland, 2023-04-14:
I am running 1.9.0RC5 with my CR-10s using the Marlin 2.65a firmware and I still see this error.

OctoPrint is running on my Desktop which is Fedora Core 34 (upgrade is being planned)
Logs attached.

[octoprint.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/11236322/octoprint.log)
[cura.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/11236323/cura.log)

commit ff0e526b (2020-09-16): Prevent trying to select the gcode file when it is already printing

Fixes #187