fieldOfView/Cura-OctoPrintPlugin #188: Have multiple printers & Octopi boxes. Will only save one API key.

starlord1980, 2020-08-15:
**Problem:**
I have added a second printer and therefore added another Raspberry Pi with OctoPrint.
In my printer list in Cura I have the two printers set up, and manually set the IPs to point to the two Octopi's.
10.0.1.41 - OctoPi Printer 1
10.0.1.43 - Octopi Printer 2

When the API key is retrieved for the OctoPi Printer 2, I connect and API key is saved.

When I select OctoPi Printer 1 - the API key has then been overwritten with the API key for Printer 2 and will no longer send commands.
If I then re-add the correct API key for Printer 1, then the API key for Printer 2 is overwritten and stops responding to commands.

**Expected Behaviour:**
OctoPi 1 Saves Its unique API key
OctoPi 2 Saves its unique API key

dataworxx, 2020-08-26:
Hi, 
i have the same problem
i choose printer 1 in cura, make settings manually ip, port api key --> close printer 1 working
i choose printer 2 in cura, make settings manually same ip, other port, other api key and immediatly the connection to the other "instance" is broken because the last api key will overwrite the api key from printer 1. 

hope you can fix it and thank you!

starlord1980, 2020-08-28:
Yep, I just uninstalled and set up Cura again and same problem, although it does seem to settle itself after a few API key retrievals of both Octopi boxes.

fieldOfView, 2020-08-29:
> it does seem to settle itself after a few API key retrievals

Hmm, those are always the tricky ones to track down. Things that don't always work. I'll try to look in to it next week.

fieldOfView, 2020-09-04:
I can confirm that working with multiple OctoPrint instances with different keys is currently iffy, and will rework that.

fieldOfView, 2020-09-07:
This [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/OctoPrintPlugin/xt1tugySfYx2tQRqUHCZ6t70kQxMD-twX8_VlFaPz16F/OctoPrintPlugin-v7.0.0-2020-09-07T09%3A31%3A13Z.curapackage?Expires=1914831079&GoogleAccessId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=BuHdHukssv4uGSIgbh8XQn0Nlf17oJsnqz1jqm2vEURh7V%2BT9xzIGZooz3yVRT6TWpBY672xRob3EPAFU748rrgDyVnvn4vB5LxKB%2FD1vpBQ862s7s%2BSfTG%2FCX%2F6LGzIhmZz%2BZ6Biy%2BPorAZqUl7ZvKtL0Qlfp6t2S%2ByZNpbLrSCkaVTsJpT8QMP4L%2Bscou9WKP2byd2tPsfedx4Ce9kD8PnwyMvL57CcDyCJaMfefhrtO484wBf5x1ZQMjlnK7hwct2XHdd%2FkALK5udaCYmoILzoPWHy9P7VibCEWWfg6ZDQjcJdlyJaZgcKRlv%2FvilQK2dpRsVGT%2BFEXlEPfm9aw%3D%3D) should fix issues with managing keys for multiple OctoPrint instances. 

Download the file and drop it onto the buildplate in Cura as if you were opening a 3d model.

fieldOfView, 2020-09-16:
Fix is available in version 3.5.17 in the Marketplace

closed 2020-09-16 by fieldOfView

focuserikfocus, 2024-05-28:
Hello everyone,

I am experiencing the same issue. I have 15 3D printers. I want to create a profile for each printer in Cura and control all of them through Cura with different IP addresses and API keys (OctoFarm does something similar, but it doesn't seem very user-friendly). The Cura plugin only allows a single IP address: localhost:5000. I wish I could enter an IP address and API key for each printer profile. I've been struggling with this for days and I'm quite a beginner. A little help would be greatly appreciated.

Thank you.