fieldOfView/Cura-OctoPrintPlugin #189: Problems requesting API key

Schmittfinger, 2020-08-31:
Hi,

I have problems to connect my Octoprint (1.4.2) to cura (tested with 4.6 and 4.7) on windows 10.
I can configure the octoprint-plugin in cura. If I click the Request... button for the API Key-field, my browser opens the octoprint-website and (after login) nothing happens.
If I click the both once again, a new browser-tab opens with octoprint and in the first browser-tap appears a windows with "Do you allow cura to access to the octoprint-API - Allow / Disallow". I clicked allow but nothing happens again.

I tested to copy/paste the "global API key" from octoprint to Octoprint-Cura-Plugin - with the result "OctoPrint is not available". What I'm doing wrong?

My cura.log is atteched.

Thanks
Sebastian
[cura.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/5150240/cura.log)

fieldOfView, 2020-09-03:
I've had a look at the logs. There are 17 cases where you pressed the `Request...` button. 16 of them end with the AppKey request being denied ("Disallow" was pressed). The last one ends with the AppKey request being accepted ("Allow" was pressed). I also see a lot of `Connection refused or timeout when trying to access OctoPrint at http://192.168.178.71:80/api/settings` errors.

Does your OctoPrint instance respond in a timely fashion when you open Connection refused or timeout when trying to access OctoPrint at http://192.168.178.71:80/api/settings?apikey=[your apikey] ?

fieldOfView, 2020-09-04:
Possibly related to #188

Schmittfinger, 2020-09-05:
Thanks for your hint.
My problem is related to #188 
I have a setup here with two printers and two octoprint-installations (for each printer).

I tested the following:
- Disabable the "automatic discover local OctoPrint instances (in the 'Connect to OctoPrint'-menu)
- Added one of my octoprint-installations manually
- Add the global api key from this octoprint-instance
- with this combination it finally works

Question at this point - can I edit the different options (see attached screenshot) once again when octoprint is already connected?

Thanks and best regards,

Sebastian
[image]

fieldOfView, 2020-09-07:
It would be great if you could test this [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/OctoPrintPlugin/xt1tugySfYx2tQRqUHCZ6t70kQxMD-twX8_VlFaPz16F/OctoPrintPlugin-v7.0.0-2020-09-07T09%3A31%3A13Z.curapackage?Expires=1914831079&GoogleAccessId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=BuHdHukssv4uGSIgbh8XQn0Nlf17oJsnqz1jqm2vEURh7V%2BT9xzIGZooz3yVRT6TWpBY672xRob3EPAFU748rrgDyVnvn4vB5LxKB%2FD1vpBQ862s7s%2BSfTG%2FCX%2F6LGzIhmZz%2BZ6Biy%2BPorAZqUl7ZvKtL0Qlfp6t2S%2ByZNpbLrSCkaVTsJpT8QMP4L%2Bscou9WKP2byd2tPsfedx4Ce9kD8PnwyMvL57CcDyCJaMfefhrtO484wBf5x1ZQMjlnK7hwct2XHdd%2FkALK5udaCYmoILzoPWHy9P7VibCEWWfg6ZDQjcJdlyJaZgcKRlv%2FvilQK2dpRsVGT%2BFEXlEPfm9aw%3D%3D), which should fix issues with managing keys for multiple OctoPrint instances. 

Download the file and drop it onto the buildplate in Cura as if you were opening a 3d model.

grantx2016, 2020-09-08:
@fieldOfView My plugin also stopped working after upgrading to Cura 4.7. I can confirm that your development snapshot fixed the issue. Thanks for the quick action on this and keep up the great work on this plugin. It's very useful.

Schmittfinger, 2020-09-08:
@fieldOfView Today I upgrade my cura to version 4.7.1 
Unfortunatly with this version I had the same issues again - with the marketplace-plugin-version and the development snapshot.
I hope I have more time for testing this weekend.

Thanks for your help.

Best regards,
Sebastian

fieldOfView, 2020-09-08:
A new day, a new [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/OctoPrintPlugin/xt1tugySfYx2tQRqUHCZ6t70kQxMD-twX8_VlFaPz16F/OctoPrintPlugin-v7.0.0-2020-09-08T07%3A59%3A33Z.curapackage?Expires=1914911977&GoogleAccessId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=aYadWlWJnMKvWItp0PXy4PU5iyRj6n5gy%2BSZ86sJfC7vCzoHKesOeUIQJzsKqXB4B46rBCGzTBv1a5WXXuoXFtQDWMUGG%2FbwzApK4qUhxHf%2F3BaC6%2B6pqKY71x50y4itXfQ3lIgiStJrxMhMvfEPiDgxsrWjAsXNw27OqN5SCljR2U0HvFEZy0yTrz5z%2B%2BCSEbOqf8IoJuRIBy9rJzeNuVIP9zkHGj7Ih4qghylG5G5FxjCEdkYtkXMdnsdBKYPIsH%2BqU2T2F8UbUQN%2FSY%2F4O8Oa9XecM7TgmUwVmsTOFYb8fu17Gtkh9oPuNSe6WEBvXcpk2za6svpxcON3k%2BSNWA%3D%3D). I managed to break the appkey workflow (the `Request...` button) in the previous snapshot, unless you were very quick.

Schmittfinger, 2020-09-15:
Thanks for your new development build. 
It works for me, too. 
Until now both of my printers had a seperate octoprint-installation each on a raspi 1.
Some of my problems with the connection between cura and octoprint seems to be the little/bad performance of the raspi 1.
Today I connected one of my printers to a raspi3 for testing (had issues with camera).
I tried to connect cura to the octoprint on the raspi3 and it worked suddenly.

Best regards,
Sebastian

fieldOfView, 2020-09-16:
Fix is available in version 3.5.17 in the Marketplace

closed 2020-09-16 by fieldOfView