fieldOfView/Cura-OctoPrintPlugin #190: Crash from Cura 4+ in Monitor Page

jackopoz, 2020-09-17:
To reproduce: connect to printer which is printing or start print, go to Monitor Page, activate another printer that is not connected to OctoPrint, CRASH!
Also can be reproduce when printer is offline

```
Fatal Python error: Segmentation fault

Thread 0x00007fbbbd0ab640 (most recent call first):
  File "/tmp/.mount_runtim58dmJP/usr/bin/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 86 in _updateThread
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fbbbd8ac640 (most recent call first):
  File "/tmp/.mount_runtim58dmJP/usr/bin/plugins/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fbbbe0ad640 (most recent call first):
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 297 in wait
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 549 in wait
  File "/tmp/.mount_runtim58dmJP/usr/bin/plugins/plugins/UM3NetworkPrinting/src/Network/ZeroConfClient.py", line 68 in _handleOnServiceChangedRequests
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fbbbf7fe640 (most recent call first):
  File "/build/4.3/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 150 in _storeStderrToLogThread
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fbbbffff640 (most recent call first):
  File "/build/4.3/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 138 in _storeOutputToLogThread
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fbbe67fc640 (most recent call first):
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 293 in wait
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/build/4.3/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/build/4.3/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 112 in run
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fbbe6ffd640 (most recent call first):
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 293 in wait
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/build/4.3/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/build/4.3/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 112 in run
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fbbe77fe640 (most recent call first):
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 293 in wait
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/build/4.3/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/build/4.3/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 112 in run
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fbbe7fff640 (most recent call first):
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 293 in wait
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/build/4.3/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/build/4.3/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 112 in run
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.3/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Current thread 0x00007fbc1324a740 (most recent call first):
  File "/build/4.3/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 341 in exec_
  File "/build/4.3/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 800 in run
  File "/build/4.3/build/inst/bin/cura", line 177 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "/build/env/4.3/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "/build/env/4.3/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
```

after some debug have found that it crashes after plugin calls `removeOutputDevice`

fieldOfView, 2020-09-21:
A segmentation fault is a bit hard to track down because it is not a crash in "my code" but in the frameworks we use. It could be indicative of something that is not properly disposed of in either Cura or my plugin. I will try to reproduce this crash.

jackopoz, 2020-09-21:
After some lengthy investigation we have determined the cause of the SIGSEGV crash in Cura code
The problem in applying styles in `Repeater` [here](https://github.com/Ultimaker/Cura/blob/c654718d8f7d3ac215c7f09bb513323955ac000e/resources/qml/PrinterOutput/ManualPrinterControl.qml#L268)

rewriting without `Repeater` have fixed that crash

> A segmentation fault is a bit hard to track down because it is not a crash in "my code" but in the frameworks we use. It could be indicative of something that is not properly disposed of in either Cura or my plugin. I will try to reproduce this crash.