fieldOfView/Cura-OctoPrintPlugin #191: Can't enable PSU Control

TScalzott, 2020-09-19:
I have Octopi running with PSU Control plugin, which functions well from OctoPrint.  
In Cura, the option to "turn on printer with..." is grayed out.

closed 2020-09-19 by TScalzott

yasars, 2021-12-05:
same too any fixx?

fieldOfView, 2021-12-05:
@yasars, this issue was closed by the user, because he figured out the "fix" by himself. You need to have the option the start printing selected. If you don't want to start printing after upload, then why would the printer need to be turned on?

If that does not fix your issue, please open a new issue with more details about what your problem is. Perhaps add a screenshot. You may also want to have a look at https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/232

yasars, 2021-12-05:
thats help thank you :) 👍