fieldOfView/Cura-OctoPrintPlugin #192: 'NoneType' has no object 'state'

sdaitzman, 2020-09-20:
[image]

This sometimes happens when I submit a print. Is there any other useful debug info I can provide?

Most recent versions of Cura and Octoprint are running.

fieldOfView, 2020-09-21:
A cura.log would be helpful; see Help -> Show configuration files. It would be nice if you could indicate an approximate time/dat of this happening, so I can find that moment back in the logs.

fieldOfView, 2020-10-01:
@sdaitzman I cannot fix this without further information requested above.

sdaitzman, 2020-10-01:
@fieldOfView thanks for the reminder! I dug more into the issue, and it turns out the true issue is that my RasPi's WiFi driver is crashing, knocking it off the local net.

Perhaps this is expected behavior, then, for when the Octoprint instance disappears? If so, I guess this issue is closed or repurposed as a feature request (to display a warning that the Octoprint instance was not found, rather than this error message).

fieldOfView, 2020-10-02:
> Perhaps this is expected behavior, then, for when the Octoprint instance disappears?

The error message is not expected behavior. The upload failing when the OctoPrint instance disappears would interrupt things through. Thanks for this extra bit of information,

fieldOfView, 2020-10-23:
For reference: https://community.ultimaker.com/topic/34388-nonetype-object-has-no-attribute-close-in-cura-71/?do=findComment&comment=274633

fieldOfView, 2020-11-15:
This should be fixed in master

fieldOfView, 2020-12-29:
A new version including a fix for this issue is available from the Marketplace.

closed 2020-12-29 by fieldOfView