fieldOfView/Cura-OctoPrintPlugin #194: Upload to OctoPrint Slow After Upgrade to 3.5.17

rebeltaz, 2020-09-24:
Ever since upgrading the Cura 4.7.0 OctoPrint Connect plugin to 3.5.17, uploading the gcode to OctoPrint is taking **forever**.... well, like five minutes anyway. It wasn't this slow before the upgrade. Is there anyway to downgrade back to the previous version?

fieldOfView, 2020-11-15:
I don't see how the recent changes in the OctoPrint Connection plugin could reduce the upload speed.

There is currently no official download of previous versions of the plugin. You can download the source of version 3.5.16 here:
https://github.com/fieldOfView/Cura-OctoPrintPlugin/archive/8affa8aa9796cb37129d3b7222fff03f86c936cd.zip

To install, go to Help -> Show configuration folder and locate the `plugins` folder. Replace the files from `OctoPrintPlugin\OctoPrintPlugin` with the files from the zip archive. You will have to rename the folder inside the zip to match the file structure of what was already there.

fieldOfView, 2021-07-06:
Did installing the older version fix the issue for you?

fieldOfView, 2021-11-09:
Closing due to inactivity

closed 2021-11-09 by fieldOfView