fieldOfView/Cura-OctoPrintPlugin #195: Octoprint plugin does not honour Select print job after uploading

MaxwellSmartAgent-86, 2020-09-24:
The checkbox Select print job after uploading seems to be completely ignored.

This has been happening since I installed Cura 4.7.0.  When I send a job to the printer, I receive a message (warning?) that the job could not be selected as the printer is printing.

The job still gets uploaded fine so it's not a big deal but I just find it a little annoying.  Of course, if the printer is idle obviously I don't see that message.

I have NOT checked to see if any other checkboxes in the settings are similarly affected.

fieldOfView, 2020-09-24:
Sounds related to https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/187

I'll look into it.

NoHow81, 2020-10-21:
I can confirm this behavior with Cura 4.4 and 4.7

fieldOfView, 2020-11-15:
Should be fixed in 8354ce0

fieldOfView, 2020-12-29:
A new version including a fix for this issue is available from the Marketplace.

commit 8354ce0e (2020-11-15): Fix handling option to select model after uploading

Fixes #195
closed 2020-12-29 by fieldOfView