fieldOfView/Cura-OctoPrintPlugin #197: Send to Octoprint without Printing

Metathesus, 2020-09-25:
Hello,
Wonderfull "plugin" that help a lot. Thank for it!
As an enhancement, Is it possible to add something like "Send GCode octoprint" without starting the print? 
So we can start the print from another device later or send files during a print?

Turiok, 2020-10-01:
Hi,

The plugin already do that.
[image]

Disable the "Start printing job after uploading" in the manage printers configuration panel for octoprint

Kind regards
Turiok,

fieldOfView, 2020-10-01:
As @Turiok already said, that functionality is already available. Also a duplicate of #129, which is still under consideration. I am considering adding a second "output" labeled "Save to OctoPrint" in addition to the current "Print with OctoPrint".

closed 2020-10-01 by fieldOfView