fieldOfView/Cura-OctoPrintPlugin #198: [FR] Link Octoprint instance to Printer Profile

CRCinAU, 2020-09-30:
I have two different pi installs with two printers - an Ender 3 Pro and Ender 3 V2.

I have two printer profiles within Cura:
[image]

I set up two connections to Octoprint:
[image]

However I cannot link each octoprint instance to a printer profile.

This means I have to manually connect to a specific instance of OctoPrint depending on which printer I want to use.

It would be great if I could link an Octoprint instance with a printer profile to allow me to automatically switch between Octoprint installs by changing the printer profile in Cura.

fieldOfView, 2020-10-16:
> I have to manually connect to a specific instance of OctoPrint depending on which printer I want to use.

This should not be the case. Both 'Printers' in Cura should remember which 'OctoPrint instance' they are connected to.

CRCinAU, 2020-10-18:
You know, I think you might be right - I was basing my observations on the highlighted printer in the OctoPrint plugins UI didn't select the printer that was associated with the printer definition - as it seemed to not exist, or not highlight the correct one.

How does it save the OctoPrint instance based on the printer definition? Is it the last one you hit "Connect" to whist in a particular definition?

fieldOfView, 2020-11-15:
> How does it save the OctoPrint instance based on the printer definition?

The plugin does not save based on the printer definition; a printer definition defines the type of printer (eg the Ender 3). If you "Add a printer", Cura creates a printer instance with that definition. You can add multiple printer instances with the same printer definition (eg if you have multiple Ender 3 printers).

The plugin saves a reference to the selected OctoPrint instance via the Connect to OctoPrint dialog when you press the connect button. You had a printer instance "activated" when you opened the Connect to OctoPrint button. The OctoPrint instance gets linked to that printer instance. If you want to link an OctoPrint instance to a different printer instance, you have to activate that printer instance first.

fieldOfView, 2020-11-15:
Let me know if the above explanation does not make it clear enough. As far as I can tell, this is working.

closed 2020-11-15 by fieldOfView