fieldOfView/Cura-OctoPrintPlugin #200: Klipper not working

Firestorm7893, 2020-10-06:
From what i understand, if you have a printer with the klipper firmware this plugin won't work. I think the reason is that klipper doesn't connect trough serial and replaces the connect button on the octoprint interface, so cura-octoprint when tries to connect to the printer "pushes" the old connect button and not the klipper one, causing the serial errors i'm seeing.
Is there anyway to add support for klipper? That would be awesome!

fieldOfView, 2020-10-06:
> cura-octoprint when tries to connect to the printer "pushes" the old connect button

Cura does not instruct OctoPrint to try to connect to the printer, unless you have the option "Connect to printer before sending printjob" selected. Do you have that option selected? 

> causing the serial errors i'm seeing.

Could you tell me what serial errors you are seeing?

Firestorm7893, 2020-10-07:
Yes i have the "Connect to printer before sending printjob" option selected

Ok i updated the klipper plugin and now i don't get the errors with serial (unfortunately i don't remembere what those were, if they happen again i will make a screenshot of them). 
But still, the cura plugin gets stuck waiting for octoprint to connect. 
[image]
This time i can manually click the connect button and then the print will start, so that's a plus :)
If you need any info i'll be glad to provide it to you.

fieldOfView, 2021-01-07:
While it is true that the OctoKlipper plugin can (optionally?) replace the Connection sidebar pane, including the Connect button with a modified one, my plugin does not care about the button; it uses the API to ask OctoPrint to connect to the printer. I can't find an API to connect to Klipper in the OctoKlipper plugin, so your request may not be possible without changes to the OctoKlipper plugin.

lixxbox, 2021-01-08:
Hi,
meanwhile many people use Klipper in conjunction with Moonraker (Klipper API) and an associated web interface (Mainsail, Fluidd, KlipperScreen, ...). This has the advantage that it runs much more performant than OctoPrint.

Moonraker allows file upload through an "Octoprint legacy upload API", which works e.g. with PrusaSlicer. 
Maybe it would be possible to implement this optionally?

Some more info:
https://github.com/Arksine/moonraker/blob/master/docs/web_api.md#file-upload

stenc55, 2021-02-08:
Hi. I will borrow this thread to support lixxbox request. I am able to connect to Octoprint, but when I try to save file on Octo, I have response "Can't store printjob on printer SD card"
As I understand, plugin will try to save file on printer's SD card, but that does not exist. Everything printer does is calculated by Octoprint and Klipper, printer only get finished commands to perform whatever it has to do, thus all files are on Octoprint. Gcodes are stored in /home/pi/.octoprint/uploads. I would not know where Moonraker/Mainsail stores them.
Maybe some code "if exists save here, else save there?
Regards

fieldOfView, 2021-02-08:
> As I understand, plugin will try to save file on printer's SD card, but that does not exist.

Then why did you select the option to store the printjob on the printer SD card? The message you mention only shows when that option is selected in the "Connect to OctoPrint" dialog, and it is not selected by default.

Syco54645, 2021-03-19:
This plugin nearly works perfectly with klipper. The only issue is that the print does not start upon upload. I have "Start print job after uploading" selected yet all it does is upload. @fieldOfView if there is a bug here that would be it I would think.

For users having issues uncheck "Automatically discover local OctoPrint instances" and add the printer using either the ip or the zeroconf dns. Under "API Key" just enter random characters (i have zx) and it will connect fine.

fieldOfView, 2021-03-19:
Please see this issue, and how it was resolved:
https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/227

closed 2021-03-19 by fieldOfView