fieldOfView/Cura-OctoPrintPlugin #203: [Request] Add plugin comment to uploaded gcode

FormerLurker, 2020-11-10:
Hi!  I was wondering if you would be willing to add a single comment to the top of files uploaded to OctoPrint that could be used to indicate that the file was uploaded via your plugin.  Something like this maybe:

```
; Uploaded by the Cura-OctoPrintPlugin
```

This would allow my plugin, ArcWelder, to detect that a file was uploaded via your plugin, and to process the file accordingly.  A bonus would be to also add an indication that the print will be auto-started:

```
; Uploaded by the Cura-OctoPrintPlugin - Print After Upload
```

Though that is not as important.  Anyway, thank you for considering this!

fieldOfView, 2020-11-13:
> This would allow my plugin, ArcWelder, to detect that a file was uploaded via your plugin,

What would be the difference between uploading with my plugin and uploading manually?

Can my plugin somehow see that your plugin is done processing the gcode file? Then it can handle the "print after upload" by itself, similarly how it waits for the PrintTimeGenius plugin to do its processing.

fieldOfView, 2020-11-13:
On a totally unrelated note, a Cura plugin that implements your ArcWelder as a postprocessor is still on my todo list.

FormerLurker, 2020-11-13:
> Then it can handle the "print after upload" by itself, similarly how it waits for the PrintTimeGenius plugin to do its processing.

PrintTimeGenius uses the gcode analysis hook I believe.  I had a lot of trouble with that method and abandoned it eventually since ArcWelder can both modify the source file and create a new file with a pre/postfix.  However, I'm open to any suggestions for how to accomplish this.

FYI, I've added the ability to add settings for arc-welder directly to the gcode file.  Things like 'print after processing' and conversion settings.  I've got some capability to detect slicer uploads if the gcode file contains any unique comments indicating where it came from.

Regarding the postprocessor, someone sent me a py file that actually works with the postprocessing plugin.  I'm going to parallelize it so that it can slice multiple layers at the same time, which should be fast enough to make it work like a regular postprocessor.  It would still be nice for it to be a full blown plugin though.

fieldOfView, 2020-11-15:
I have no problem with adding a comment to the uploaded gcode to identify the plugin, but I find adding a "print after upload" flag to the gcode an ugly hack. That flag only ever makes sense right after uploading, but remains in the gcode file on the OctoPrint instance.

I would rather poll an endpoint to see if ArcWelderPlugin is done processing the file, or tell ArcWelderPlugin to start the print after processing with an API call to the ArcWelderPlugin.

FormerLurker, 2020-11-16:
Maybe I can just provide an api call?  Something like /arcwelder/printAfterProcessing?  Internally I would check the processing queue to see if the provided path is in there, and if it is it will mark it to print after processing.  If not, it could return is_queued=false or 404 if the plugin is not found.

If something like this will work, I'll add the call and will provide a definition.

fieldOfView, 2021-11-09:
I forgot to respond to your last reply.

> If something like this will work, I'll add the call and will provide a definition.

That would work.