fieldOfView/Cura-OctoPrintPlugin #205: Cura 4.8.0 stabil can not find octoprint instances

00monter00, 2020-11-14:
Thank you for making Octoprnt connection app. ;)
Application version
4.8.0 stabile

Platform
win10 64 pro

Printer
ender 3 pro

Reproduction steps

update all apps also ostoprint connection to 3.5.1.7
Screenshot(s)
https://i.postimg.cc/m26wLZKZ/octo.jpg

Additional information
I have issue now also with 4.8.0 beta but not with 4.6.1 (still i have). I'm not only one with this issue. I erase 4.7.1 and 4.8 beta with instalation of 4.8.0. (before instaling 4.8.0 stabil all version have not any issue with octo app) Now i istalled one more time 4.8 beta and also issue exist in 4.8 beta.

P.S. Please forgive me if something is wrong, I am not a specialist in these matters and I am old, although I am trying.

fieldOfView, 2020-11-14:
What version of OctoPrint do you use?

What version of the OctoPrint Connection plugin do you use in Cura 4.6?

00monter00, 2020-11-14:
In 4.6.1 3.5.12 and version 1.4.2
In 4.8 beta and 4.8.0 3.5.17
 (i hope this is what you ask)

fieldOfView, 2020-11-15:
Can you give me the output of the following:
* press windows-key and `r`
* type `cmd` followed by enter
* in the Command Line window that opens, type `dns-sd -B` followed by enter
* let the command run for a couple of seconds

00monter00, 2020-11-15:
C:\Users\Demon>dns-sd -B
Browsing for _http._tcp
Timestamp     A/R Flags if Domain                    Service Type              Instance Name
15:21:44.518  Add     2 11 local.                    _http._tcp.               vuzero
15:21:44.619  Add     2 11 local.                    _http._tcp.               OctoPrint instance on octopi
15:21:44.722  Add     2 11 local.                    _http._tcp.               Brother DCP-J4110DW

fieldOfView, 2020-11-15:
It would be great if you could do some tests for me.
* Download this zip archive: https://github.com/jstasiak/python-zeroconf/archive/2316027e5e96d8f10fae7607da5b72a9bab819fc.zip
* Get to Help -> Show configuration folder
* Navigate to plugins/OctoPrintPlugin/OctoPrintPlugin/python-zeroconf
* Replace the files there with the files in the zip archive (overwriting the files in the same location as they are)

00monter00, 2020-11-15:
I did as you wrote, it is still as it was.

fieldOfView, 2020-11-15:
Can you please be painfully clear in what you say?

> it is still as it was.

Do you mean that the issue is not solved by replacing those files, or is the situation how it was with version 3.5.12 of the plugin?

00monter00, 2020-11-15:
Sorry, poor english.
4.6.1 all ok.
4.8.0 issue still exists
that is, after replacing the files in the folder, nothing has changed regarding the connection

fieldOfView, 2020-11-15:
Thanks next thing to try is to make a backup of your configuration folder for Cura 4.6.1, and then update the plugin in Cura 4.6.1. If discovery still works in Cura 4.6.1 with the newer version of the plugin, the error is not in my code so I have to look what changed in Cura. 

Ofcourse the flipside of that test is that you can also try to see if copying the entire plugins/OctoPrintPlugin folder from the 4.6.1 configuration folder over the version you have in the 4.8 configuration folder. If that makes your 4.8 install see OctoPrint instances, then the fault is in my code.

00monter00, 2020-11-15:
So ... I applied the second part of your test and copied the OctoPrintPlugin folder from 4.6.1 to 4.8.0 latest and it is still the same issue (with 3.5.12), also after updating to 3.5.17 nothing has changed. Sorry but I won't take the risk of copying files to 4.6.1 because it's the only working slicer for me at the moment and a lot of profiles.
I understand that this has exhausted all possible help on your part. Thank you so much for your patience and your dedication to my problem. I cordially greet you.
P.S. If you realy need test on 4.6.1 with updated app...i will try ;)

fieldOfView, 2020-11-16:
I understand your hesitation. It should be safe to backup your configuration folder since that has al your profiles in it. If it doesn't work anymore, you should just be able to put back the backup. But I can't say with 100% certainty there's nothing going to change outside the configuration folder; I'm only 99% certain.

You could try Cura 4.7, which is still available here: https://github.com/Ultimaker/Cura/releases/tag/4.7.1

00monter00, 2020-11-16:
An interesting fact is that after reinstalling 4.7.1 cura behaves the same as in version 4.8.0 latest ... and before removing I was connected to octoprint but now  can not find octoprint instances...

fieldOfView, 2021-11-09:
The next version of the plugin will use an updated version of ZeroConf that may help with this issue.

Please help me test this feature by downloading this [development snapshot](https://storage.googleapis.com/um-cloud-production-cura-packages/v1/packages/OctoPrintPlugin/FF_CBKkntJes5ApbMoHWqgklxPoLeubdesoMFX62xHHo/OctoPrintPlugin-v7.0.0-2021-11-09T17%3A14%3A47Z.curapackage?X-Goog-Algorithm=GOOG4-RSA-SHA256&X-Goog-Credential=cura-packages-api%40um-cloud-production.iam.gserviceaccount.com%2F20211109%2Fauto%2Fstorage%2Fgoog4_request&X-Goog-Date=20211109T171503Z&X-Goog-Expires=604800&X-Goog-SignedHeaders=host&X-Goog-Signature=68c6272055900f6c5a7db3a029fc54ba8adb7e90b0a3bd0ac52b071b036aa76b04af2b5a0aab494f6c1c93aadf13c3fcb8bce93635439121b1ad52d64c053cff767bf336edec6e2cb40a3f10b72d871f67dc983f351e842ae7f7da6d08d93cf9aa99ffc2f15abc9f5414dc295b08c42d3973c924fd4ddc8f1b05ead5562ee471effa9890234e09003eb244fbfe47fde9a295fdb21eae4bfcb7fc59169761747e8f4a102fde5232c91f0f0242fa43a1de2f5944c9045bb95fa4469cb3fcb3c41c2ed5e27f74d686c5ca2da373e95ddedfb146f6fbb874b7c9f4fc8617fabd2d3ad6b91f9be68183d7e712f781aac7dbe17e00df479378392513976c2d5b412548) and dropping it on the Cura buildplate to install it. The option can be found in the Connect to OctoPrint dialog.