fieldOfView/Cura-OctoPrintPlugin #209: Cura 4.8 crashes when entering the API Key

Fabmaszter, 2020-11-22:
I have a brand new **Octoprint Version 1.4.2** installation and a freshly installed **Cura 4.8**.
The only thing attached to the Raspi is the original RaspiCam (I also tried without the cam, same result).
When I enter the API in the "Connect to OctoPrint" screen (it also does not matter what I enter, even if it's just a single digit, same result) Cura crashes with this error:

[image]

**Here's the full error trace:**
Traceback (most recent call last):
  File "C:\Users\Fabmaszter\AppData\Roaming\cura\4.8\plugins\OctoPrintPlugin\OctoPrintPlugin\DiscoverOctoPrintAction.py", line 538, in _onRequestFinished
    self._settings_instance.parseSettingsData(json_data)
  File "C:\Users\Fabmaszter\AppData\Roaming\cura\4.8\plugins\OctoPrintPlugin\OctoPrintPlugin\OctoPrintOutputDevice.py", line 1202, in parseSettingsData
    self._webcams_model.deserialise(webcam_data)
UnboundLocalError: local variable 'webcam_data' referenced before assignment

**And the Log:**
Thread 0x0000098c (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.8.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00003978 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1434 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000038a4 (most recent call first):
  File "threading.py", line 297 in wait
  File "threading.py", line 549 in wait
  File "C:\Program Files\Ultimaker Cura 4.8.0\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000010c4 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1262 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001a98 (most recent call first):
  File "zeroconf\__init__.py", line 1183 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001ba0 (most recent call first):
  File "threading.py", line 297 in wait
  File "C:\Users\Fabmaszter\AppData\Roaming\cura\4.8\plugins\OctoPrintPlugin\OctoPrintPlugin\python-zeroconf\zeroconf\__init__.py", line 2184 in wait
  File "C:\Users\Fabmaszter\AppData\Roaming\cura\4.8\plugins\OctoPrintPlugin\OctoPrintPlugin\python-zeroconf\zeroconf\__init__.py", line 1496 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000033bc (most recent call first):
  File "threading.py", line 297 in wait
  File "C:\Users\Fabmaszter\AppData\Roaming\cura\4.8\plugins\OctoPrintPlugin\OctoPrintPlugin\python-zeroconf\zeroconf\__init__.py", line 2184 in wait
  File "C:\Users\Fabmaszter\AppData\Roaming\cura\4.8\plugins\OctoPrintPlugin\OctoPrintPlugin\python-zeroconf\zeroconf\__init__.py", line 1315 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000c74 (most recent call first):
  File "C:\Users\Fabmaszter\AppData\Roaming\cura\4.8\plugins\OctoPrintPlugin\OctoPrintPlugin\python-zeroconf\zeroconf\__init__.py", line 1221 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000047c (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.8.0\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000324c (most recent call first):
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 165 in _storeStderrToLogThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00003740 (most recent call first):
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 153 in _storeOutputToLogThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000030fc (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000cb0 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001f3c (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000182c (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Current thread 0x00003808 (most recent call first):
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 396 in _logInfoWidget
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 182 in _createDialog
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 86 in __init__
  File "X:\4.8-exe\build\inst\bin\cura_app.py", line 167 in exceptHook
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 378 in exec_
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 847 in run
  File "X:\4.8-exe\build\inst\bin\cura_app.py", line 235 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load

**Here are the other log files (stdout.log is empty):**
[stderr.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/5580530/stderr.log)
[cura.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/5580532/cura.log)

Before it was all working fine, then my Raspi died and when setting up the new one this whole situation came up. I completely removed Cura and reinstalled it without any change.

Fabmaszter, 2020-11-22:
Oh wow! Turns out the problem was that I switched Octoprint to Port 81 because I am also running PiHole on the same Raspi and the auto detect recognizes it as port 80. When I manually add it with the correct port it works!

fieldOfView, 2020-11-23:
Even if you have misconfigured your OctoPi instance, the plugin should not cause a crash.

To fix the port that OctoPrint advertises via bonjour, see your config.yaml. By default, octopi configures that as port 80:
https://github.com/guysoft/OctoPi/blob/devel/src/modules/octopi/filesystem/home/pi/.octoprint/config.yaml#L9
If you change the port OctoPrint is hosted on, also change this variable in your `/home/pi/.octoprint/config.yaml`

Fabmaszter, 2020-11-23:
Alright, thanks! I changed that variable to 81 and now also the auto-discover works!

fieldOfView, 2020-11-23:
The actual bug (crashing on the error `UnboundLocalError: local variable 'webcam_data' referenced before assignment`) has already been fixed here: 34cae09. This fix will be in the next version of the plugin.

fieldOfView, 2020-12-29:
A new version including a fix for this issue is available from the Marketplace.

closed 2020-12-29 by fieldOfView