fieldOfView/Cura-OctoPrintPlugin #212: Can't connect Cura to Octoprint 1.5.1

ctwise, 2020-12-06:
First time using Octoprint and first time using Cura connector. I created an API key for my user, brought up Cura, brought up Connect to OctoPrint and entered the info for the octoprint server. I cliked on the 'Request...' button and a pop-up shows in Octoprint asking for approval. I press 'allow'. But on the Cura side it never changes from 'OctoPrint is not available'.

I put Wireshark on it and that just confuses me more. At either see '403 Forbidden' for each request or I see 202 waiting for approval messages.

I also never see my user in the 'User name' field under the API Key field.

Suggestions?

ctwise, 2020-12-07:
I've continued to dig and the plugin doesn't ask for a username and password even though access control is turned on in Octoprint. I tried filling in the username in the proxy settings but that didn't work. The User name field just stays blank.

I'm able to access the http://octopi:80/api/settings from my browser but I see this in the Cura logs:

2020-12-06 20:13:36,088 - WARNING - [MainThread] OctoPrintPlugin.DiscoverOctoPrintAction._onRequestFailed [442]: Connection refused or timeout when trying to access OctoPrint at http://octopi:80/api/settings

ctwise, 2020-12-07:
Discovery doesn't work because the printer is on a different network segment.

fieldOfView, 2020-12-07:
I think this is a duplicate of #211

fieldOfView, 2020-12-07:
> the plugin doesn't ask for a username and password

This is correct. The plugin uses keys instead of a username/password. This key has a user connected to it.

ctwise, 2020-12-07:
I don’t know if it’s the same problem or not. It’s not due to a timeout
according to the logs.

fieldOfView, 2020-12-07:
The only bit of log that you posted says:

OctoPrintPlugin.DiscoverOctoPrintAction._onRequestFailed [442]: **Connection refused or timeout** when trying to access OctoPrint at http://octopi:80/api/settings

ctwise, 2020-12-07:
That’s a very valid point. And I verified that you are correct. I increased
the timeout and it's working now. You can mark this as a duplicate of 211.
Thanks for setting me straight. :-)

closed 2020-12-07 by fieldOfView