fieldOfView/Cura-OctoPrintPlugin #215: Turn on printer without starting print job?

RafaelSchridi, 2020-12-12:
Currently the plugin can **only** turn on the printer if you've select the "start print job after uploading". 
But I only need the plugin to turn on my printer, So when I disable the start print option the power control is greyed out and doesn't work anymore.

Can we get these settings be independent of each other?
[image]

lsaranto, 2021-01-07:
I would prefer this too.

Simonlebucheron, 2022-09-02:
Is this issue resolved ? Because I have the same need and the PSU control option "Turn on prior to printing after API upload" should have solve this. 
I am trying to understand where is my problem. 

As getaround for now, to abort printing but switching on the PSU, I have enabled "start print job after uploading" in the curaoctoprint plugin and I quickly click the "queue" button on the pop up windows after clicking on the print with octoprint button, see below:

[image]