fieldOfView/Cura-OctoPrintPlugin #217: TP link smartplug not detected

epalzeolithe, 2021-01-01:
Here's my config
MacOS 11.1
Cura 4.8.0
Octoprint 1.5.2
Octoprint smartplug 0.9.26
Cura OctoprintPlugin Fieldofview 3.5.18

Well the plugin works well, except, i doesnt detect the tplink plugin in octoprint so i cant the select it in configuration of octoprintplugin. I need to manually power on the printer on octoprint before. After that, everything works well for sending jobs and monitoring the printer

Do i need a special configuration ?

epalzeolithe, 2021-01-01:
SOLVED 👍 
i read the python code, and understand it needs a label in the tp link configuration, otherwise it doesn't find the plug.

Perhaps we could add some code to define a generic label, if no label is defined :)

```
        common_api_plugins = [
            ("tplinksmartplug", "TP-Link Smartplug", []), # ip
            ("orvibos20", "Orvibo S20", []), # ip
            ("wemoswitch", "Wemo Switch", []), # ip
            ("tuyasmartplug", "Tuya Smartplug", []), # label
            ("domoticz", "Domoticz", ["idx", "username", "password"]), # ip, idx, username, password
            ("tasmota", "Tasmota", ["idx"]), # ip, idx, username, password, backlog_delay
        ]  # type: List[Tuple[str, str, List[str]]]
        for (plugin_id, plugin_name, additional_data) in common_api_plugins:
            if plugin_id in plugin_data and "arrSmartplugs" in plugin_data[plugin_id]:
                for plug_data in plugin_data[plugin_id]["arrSmartplugs"]:
                    if plug_data["ip"] and plug_data["label"]:
                        plug = OrderedDict([
                            ("plugin", plugin_id),
                            ("name", ("%s (%s)" % (plug_data["label"], plugin_name))),
                            ("label", plug_data["label"]),
                            ("ip", plug_data["ip"])
                        ])
                        for key in additional_data:
                            plug[key] = plug_data.get(key, "")
                        self._available_plugs[self._createPlugId(plug)] = plug

```

fieldOfView, 2021-01-01:
You should not need special configuration inside Cura to detect the smartplug, but the smartplug needs to be properly configured in OctoPrint. Specifically, the (IP) address of the plug must be set. But the plugin would not work in OctoPrint otherwise.

epalzeolithe, 2021-01-02:
Thank you, the label was missing :)

fieldOfView, 2021-01-02:
I'm keeping this open, because I want to make the plugin not require a label.

commit 3dd63892 (2021-01-07): Allow plugs without a label

Fixes #217
closed 2021-12-21 by fieldOfView