fieldOfView/Cura-OctoPrintPlugin #218: [Request] Adjust filename before sending to Octoprint

OllisGit, 2021-01-10:
Hi,

I am sending my models from Fusion 360 to Cura, but unfortunately Fusion sends something like this to Cura:
```
/private/var/folders/yr/m1wj20qn5lqfhnz403tqqrch0000gn/T/Neutron/c608df10-abaf-4ac1-a456-8b95317b586f.stl
```
So, each time I sliced my model and sent it to OP I forgot to rename it to the correct filename (e.g. ```A10_c608df10-abaf-4ac1-a456-8b95317b586f -> EngineHolder```)

It would be really really helpful if the Cura-Plugin could have an option for adjusting the filename before sending. Then OP and its plugins (like PrintHistory) could show a human readable filename.

Maybe something like this.
- Popup Dialog before sending, showing  the current filename which could be changed
- In the plugin-settings there is an option to disable this popup (like current implementation)

What do you thing?

cr45hmurphy, 2021-02-04:
I would also like to see this feature be implemented.

fieldOfView, 2021-02-05:
> It would be really really helpful if the Cura-Plugin could have an option for adjusting the filename before sending. 

Cura already has that. In the lower left corner of the screen, there is a printjob name with a pencil icon next to it. Click the pencil and type a new name.

I'm planning to include an optional popup too, so I'll leave this open in the mean time.

cr45hmurphy, 2021-02-05:
> Cura already has that. In the lower left corner of the screen, there is a printjob name with a pencil icon next to it. Click the pencil and type a new name.
>
> I'm planning to include an optional [...]

I was actually getting on here to say just that. I found it last night, and I felt like an idiot at that point.

jsexauer, 2021-04-19:
I am also interested in this feature.  Thanks for the tip for the pencil, that's great work around.

grego33, 2021-07-17:
I was looking for the same thing.  I knew about the pencil icon but I forget to set that almost every time.

It would also be nice to be able to save to a specific folder in Octoprint.  PrusaSlicer does this.  For example if I were to slice an xyzCube it would be nice to be able to specify `tests/xyzCube` as the name so Octoprint puts it in my `tests/` folder.  Unfortunately, Cura seems to disallow the slash characters in that pencil icon field.

InnoProd, 2021-09-08:
> 
> 
> I am also interested in this feature. Thanks for the tip for the pencil, that's great work around.

I recommend @fieldOfView's [PrintJob Naming plugin](https://github.com/fieldOfView/Cura-CustomJobPrefix) in conjuction to this discussion. I realize it won't solve your Fusion360 filename issue, but it streamlines my workflow considerably over having to edit the filename in Cura every time to contain the necessary data.

kc2zgu, 2022-05-16:
I had no idea about the "pencil icon" (like most things in the Cura GUI it uses a tiny stylized picture instead of words as a label, isn't in a logical place near other controls, and has no tool tip).

I noticed this also can trigger a bug in Octoprint, coming from a different use case though. If you upload a file with the same name as an existing one (for example, multiple copies of the same model, so the file name generated from just the input file will be the same) it will use the first instance of the file for progress reporting and Gcode viewer sync. I'm now at 324% print progress with no idea how much time is left :D