fieldOfView/Cura-OctoPrintPlugin #220: Cura 4.8 + Octoprint Unable to Connect / Printer offline

bitboy85, 2021-01-12:
Hi,

i'm using Cura 4.8 and octoprint 1.5.2
when i try to print it says printer offline.

Then i open the configuration screen select "connect to octoprint" message "checking api key" appears followed by "octoprint is not available"
If i now cut and paste the excact same api key it seems to be connected, but it is still not possible to send the print job.
Closing and opening the settings window again will lead to "octoprint is not available"

I already recreated the connection and tried the "master" api key but with same result.

geNAZt, 2021-02-28:
I installed this plugin today and had a similar issue. I can't direct "Print on Octoprint" since it tells me that the printer is offline. When i enable "Connct to printer before sending printjob" it gets infinite loading on trying to connect.

If i queue from the infinite loading bar it loads the job onto the printer and i can start it from there. 

Cura Logs show this: 

```
2021-02-28 08:19:40,006 - DEBUG - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice.requestWrite [446]: Sent command to connect printer to OctoPrint with current settings
2021-02-28 08:19:40,296 - WARNING - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [862]: Encountered unexpected printjob state: Detecting serial connection
```

It seems that it doesn't like when OctoPrint has autodetect serial on when connecting to it?

fieldOfView, 2021-07-06:
@bitboy85 since you have a workaround, and no way to reproduce this reliably, I am going to close this issue for now. Let me know if you find a way to reproduce this

@geNAZt you seem to have a different issue. See #161

closed 2021-07-06 by fieldOfView