fieldOfView/Cura-OctoPrintPlugin #223: Do i need a delay to powerup the printer ?

epalzeolithe, 2021-01-24:
Hi

you do a great job !

Well, im using CURA+TPLINK in octoprint to power the printer and your plugin.

Well, it sends well the file to octoprint, and powerup the printer.

But sometimes, the print job start, sometimes not, i need to go to octoprint ti launch the print (file already loaded)

Its a bug ? Or perhaps i need some delay to powerup properly the printer, and the plugin tries to launch the print before the connexion is properly made between octoprint and the printer ?

Thanks in advance :)

fieldOfView, 2021-07-06:
I need logs from both Cura and OctoPrint, with a timestamp of where this issue occurred.

fieldOfView, 2021-11-09:
Closing this issue due to lack of info

closed 2021-11-09 by fieldOfView