fieldOfView/Cura-OctoPrintPlugin #226: Turn on printer with grayed out

ingo916, 2021-01-29:
How do I enable this feature? I have the TP-Link plugin and its working with Octoprint.
OctoPrint 1.5.3
Python 3.7.3
OctoPi 0.18.0
TP-Link Smartplug (0.9.26)

[image]

ingo916, 2021-01-29:
i had to enable start print jog after uploading, is there a way to to turn on the printer without starting the print job?

fieldOfView, 2021-02-05:
> i had to enable start print jog after uploading

Indeed, that is the design. Why would you want the plugin to turn on the printer if you don't want the plugin to start the print? If you want to manually start the print, can't you also manually power on the printer?

closed 2021-02-05 by fieldOfView