fieldOfView/Cura-OctoPrintPlugin #228: Missing 'Connect OctoPrint' Button?

Trevorlg, 2021-02-25:
Just wanted to give the OctoPrint plugin a shot. Is there an issue with Cura 4.8.0?
I have it installed and have been searching for tutorials. All I'm locating are for older versions of Cura and the UI has changed quite drastically. They all say:

Go to the add a printer area and click the Connect OctoPrint, but that is nowhere to be found in my version

Manage Printers:
[image]

Plugin:
[image]

fieldOfView, 2021-02-25:
Thanks for opening an issue.

What printer do you have? You seem to have configured Cura for an Ultimaker S5. The Ultimaker S5 cannot be controlled with OctoPrint, so it cannot be connected to OctoPrint in Cura. If you have a different printer, then make sure you select the appropriate model when adding your printer to Cura.

Trevorlg, 2021-02-25:
> Thanks for opening an issue.
>
> What printer do you have? You seem to have configured Cura for an Ultimaker S5. The Ultimaker S5 cannot be controlled with OctoPrint, so it cannot be connected to [...]

I was unable to get Cura to get past the basic setup. 

I have a SnapMaker 2.0 A350 printer. 
Connected to that is a rPi 3B+ serially connected and running OctoPrint

Cura has no option to add SnapMaker or OctoPrint in the list of printers
I cannot get to Cura's marketplace without adding a printer in their new program, so I have to pick something to get past the welcome/setup. Could not find a way to cancel out of that.

I will uninstall Cura and reinstall to see if perhaps it will work now that my Cura account has this plugin assigned. If not, at least I have show what I'm talking about

Trevorlg, 2021-02-25:
So, Uninstalled and reinstalled Cura
Welcome --> Disclaimer/agreement --> release notes --> Anonymous data -->Sign in
Sits at an add printer screen (scan, manual ip, cloud)
[image]

I have to add something to get past this view it seems and the SnapMaker as well as the OctoPrint are not options
Restarted Cura just to be sure. Same holdup

Manually:
[image]

Tried adding via cloud:
[image]

Put the local OctoPrint IP in:
[image]

That's why I added a machine (any machine) to just get out of this screen

fieldOfView, 2021-02-25:
If your printer is not listed, select either a printer similar to yours, or select the "Custom FFF Printer". DO NOT select an Ultimaker S5, nor any of the other Ultimaker printers. They come with very specific settings for those printers that will not work well for your printer at all. 

You may want to follow this guide:
https://www.youtube.com/watch?v=aLY85usal60

Trevorlg, 2021-02-25:
Returning with my updates. I see you suggested what I tried

Basically did just that, 

Completely removed Cura
Went through the wizard and set up a FFF printer
Sync with cloud to get the plugin
Restart Cura

And voila!

I will watch that setup videos as well that you shared (wish that came up in my searching for solving this)

Thanks for the effort helping me out!

closed 2021-02-25 by Trevorlg