fieldOfView/Cura-OctoPrintPlugin #229: Cura 4.8 MacOS Big Sur Octoprint latest

uhlhosting, 2021-03-12:
Hi,

I am trying to connect Cura to my printer with OctoPrint and it fails, since I do not seem to find anywhere the 3rd step mentioning in connection steps. The button to do this action:

Select "Connect to OctoPrint" on the Printers pane of the preferences. It is not visible anywhere in the Cura app.

Can you explain what I am doing wrong, does this only works with the Windows version of the app? Is there some plugins needed on OctoPrint for this to work?

holstvoogd, 2021-03-13:
I had the same issue on  MacOS just now, found this in the logs:

```
2021-03-13 16:16:28,664 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [462]: Plugin [OctoPrintPlugin] has been disabled. Skip loading it.
2021-03-13 16:19:52,166 - WARNING - [MainThread] OctoPrintPlugin.OctoPrintOutputDevicePlugin.<module> [49]: Supplied version of Zeroconf module imported
```

I was able to fix it by manually editing plugins.json and removing OctoPrintPlugin from the disabled plugins. On windows this should be found here: C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\plugins.json

I will probably say ``` "disabled": ["OctoPintPlugin"]```, change that to ``` "disabled": []``` and restart cura.

fieldOfView, 2021-03-13:
@holstvoogd the official way to reenable a plugin that has been disabled is via the "Installed" tab in the Marketplace, by checking the checkbox in front of the plugin name. That does not get unchecked by itself though; you must have either accidentally or purposely disabled the plugin at some point. 

I will also note that it is very strange that the second line shows up in your log when the plugin is disabled. That should not have been the case. This leads me to believe you have the plugin installed in two places or something to that effect. Have you ever tried to manually install the plugin?

@uhlhosting I don't know what you are doing wrong without a look at your logs. In your configuration folder (see Help -> Show configuration folder) there should be a file named cura.log. Please upload it somehwhere and post a link here.

uhlhosting, 2021-03-13:
Seems to work. It was somehow disabled, i enabled it, restarted Cura and started working.

holstvoogd, 2021-03-14:
@fieldOfView Ah yes, of course, I forgot to mention: 
I was not able to re-enable the plugin from there. It showed up, as disabled, but I could not click the check box.

I've got the feeling this might be due to me having it installed in a previous version, but disabled.

I do not recall ever installing it manually, but maybe I did a long time ago :)

fieldOfView, 2021-03-14:
>I was not able to re-enable the plugin from there. It showed up, as disabled, but I could not click the check box.

I would need the entire log to figure out what was going on then, but since you already figured it out somehow I guess we will never know.

closed 2021-03-14 by fieldOfView