fieldOfView/Cura-OctoPrintPlugin #230: Force ipv4 instead of auto configuring ipv6

rklasen, 2021-03-24:
Hi, thank you for this fantastic plugin!

Unfortunately, I have a problem with the ipv6 auto config. I've added my printer via its ipv4, and everything worked. Sometimes the printer is available via its ipv6, but not always (firewall rule when connected to a VPN, the ipv6 is a public address, but the firewall prohibits all incoming ipv6 connections).

Obviously I'd still like to print when I'm connected to my company VPN. The easiest solution for me would be to force the plugins to use the local 192.168.* ipv4. How can I do that? The plugin seems to randomly use the ipv6, and I can't change the ip.

Thanks!

fieldOfView, 2021-07-06:
The plugin uses the IP address that it gets, either from bonjour/zeroconf autodiscovery, or from you. If you add an instance via the "Add" dialog instead of using the autodiscovery, and give the plugin a hostname, then it is up to your system to resolve the hostname to an ip address. You could try disabling ipv6 altogether on the network interface if you never want to use ipv6.

fieldOfView, 2021-10-04:
Closing due to lack of information

closed 2021-10-04 by fieldOfView