fieldOfView/Cura-OctoPrintPlugin #232: Cannot change option Turn on printer with - unknown plug

FrancYescO, 2021-05-03:
Hello i was using both PSU Control, and Tuya Smarplug plugin on my octoprint instance, the tuya plugin was unstable so i totally removed the plugin from octoprint, i never figure out how to set the plugin to turn on via PSU Control as both options in the selectbox are disabled and i have no possibility to change, it automatically select the last option (that actually i assume is the unistalled tuya smartplug plugin)

[image]

FrancYescO, 2021-05-05:
i've solved by going to (MacOS)`/Users/user/Library/Application\ Support/cura/4.9/machine_instances/myprinter.cfg` and removing the line `octoprint_power_plug = tuyasmartplug/3D Printer/192.168.1.103` after the restart finally the `Unknown plug` option disappeared and i'm able to select PSU Control

but i think this should be considered a bug.

melyux, 2021-10-02:
I had this problem too. When you change the OctoPrint's IP address in Cura, it should update the `octoprint_power_plug` line as well. It only changes the OctoPrint address, but `octoprint_power_plug` is left with the old IP address, causing this bug.

fieldOfView, 2021-10-04:
Sofar I've only seen this reported by Mac OS users. I'll get this fixed in the next release.

fieldOfView, 2021-10-04:
This looks like a MacOS specific issue in Qt: https://forum.qt.io/topic/123101/quick-controls-1-combobox-items-unselectable-on-macos

Unfortunately the fix (changing the controls to QtQuick 2) means that the dropdown will look ugly and be unreadable on current Mac OS systems (possibly limited to M1, but I am not sure of that).

Stuck between a rock and a hard place...
I'll keep looking for solutions.

fieldOfView, 2022-04-29:
Can you check if this is still an issue with Cura 5? This post has a prerelease of the OctoPrint Connection plugin:
https://community.ultimaker.com/topic/40313-fieldofview-plugins-for-cura-50/

melyux, 2025-04-09:
Still happening in the latest Cura 5.