fieldOfView/Cura-OctoPrintPlugin #234: Octoprint doesn´t work properly with the Cura 4.9.1 version

dbombin, 2021-05-15:
Now when I print something from Octoprint, I can see the behaveour of the hotend is wrong because it makes a little stops during the printing (or dropping the speed) and mekes it a wrong printing.
When I used Cura 4.9.0 with this Octoprint 1.6.1, all the system went very well.

I have tried to print the same test without Octoprint, only from SD card, and the printing is well done. So the printer is well.
[octoprint-systeminfo-20210515130644.zip](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/6487288/octoprint-systeminfo-20210515130644.zip)

fieldOfView, 2021-05-15:
What happens when you save the gcode file to disk in Cura and then use the web interface to print? If you get the same "little stops" via the web interface, then the issue you have is not with this plugin.

dbombin, 2021-05-16:
I have tried what you say, and don't do any little stops.But I think it could be because of I have my Raspberry connected by WiFi dongle to network, but PC with cura connected via ethernet to network.And when I save the gcode to the PC and then to the SD Card through Octoprint, then the print is well done because don't use the WiFi connection.I think this is the reason of the problem.I'll try to put ethernet to raspberry and I'll say you the results.Thank youEnviado desde mi Galaxy
-------- Mensaje original --------De: Aldo Hoeben ***@***.***> Fecha: 15/5/21  22:24  (GMT+01:00) Para: fieldOfView/Cura-OctoPrintPlugin ***@***.***> Cc: dbombin ***@***.***>, Author ***@***.***> Asunto: Re: [fieldOfView/Cura-OctoPrintPlugin] Octoprint doesn´t work properly with the Cura 4.9.1 version (#234) 
What happens when you save the gcode file to disk in Cura and then use the web interface to print? If you get the same "little stops" via the web interface, then the issue you have is not with this plugin.

rogtuf, 2021-05-17:
I have been running 4.8.0 for over a year now on my 2018 Macbook Pro.  Really great software and thank you for producing it.

What was not great was updating to 4.9.1 today.  

The box on the centre screen showing the workpiece seems to have been moved further up on the display so that the top has disappeared, but more importantly, despite considerable effort, it refused to recognise my Octoprint API making the upgrade unusable for me.

Thankfully I was able to restore the PC back a day with Time Machine to 4.8.0 with which I am quite happy.

I just thought you should know that there could be an issue with 4.9.1.

1MacGeek, 2021-06-02:
I am experiencing **_NO_** issues with Cura 4.9.1, my Ender 3 v2 (running 1.0.2 firmware) and the OctoPrint Connection plug-in (v.3.5.18). Despite running Cura 4.9.1 for a few weeks with no known issues, after reading these comments I uninstalled both Cura and the plug-in from my Mac, running macOS Big Sur 11.2.3. I re-installed both Cura and OctoPrint Connection, and connected Cura to the OctoPrint without any known issues. Printing to my Ender 3 v2 is also without a known issue.

Taking it one step more, I updated my Mac to macOS 11.4, and there are still no known issues using Cura 4.9.1 and OctoPrint Connection 3.5.18. 

By weight of this evidence, there is a solid case to be made there is not a general issue with Cura 4.9.1 and this plug-in.

If one truly wishes to help the developer, then please take this to heart not as a criticism but as a guide :

1. There is no such thing as too much detail. List what printer, what firmware is on that printer, your OS, et cetera. Pixels are free, so fill up your posting with as much detailed and specific information as possible.
2. Everybody hates it, but really - unplug it all and plug it all back in. Make sure all connections are solid before reporting any trouble.
3. Likewise, uninstall and reinstall all the relevant software. Should that not be an option, check for updates.
4. Hit the forums - see if others are experiencing the same issues, or if the issue is local to your equipment.
5. RTFM - make sure there isn't something you have missed.
6. If all else fails and all other means of troubleshooting either don't exist or have proved fruitless, let the developer know while giving particular attention to point #1.

Again, this is not meant to criticize anyone for anything, but it is intending to be a help to the developer and everyone else. By way of information I will say OctoPrint Connection did not recognize my Ender 3 v2 during set-up the first time I installed it. I did uninstall the plug-in and re-install it, and everything started working as advertised. Simple troubleshooting resolved the issue in moments instead of days or weeks.

Please help take some of the burden off the developer by trying some basic troubleshooting steps before reporting an issue.

fieldOfView, 2021-07-06:
@dbombin, looking at your systeminfo bundle, it seems you are running a singlecore Raspberry Pi Model B Plus Rev 1.2. That hardware is not supported for OctoPrint.

It could be that 4.9.1 creates more detailed gcode, which only just works when you print through the web interface, but stars to be too much for your poor 700 Mhz single core when it also needs to send updates to Cura over Wifi. Either way, it is not something that I think I can change from this plugin.

closed 2021-07-06 by fieldOfView