fieldOfView/Cura-OctoPrintPlugin #237: Connecting to octoeverywhere error

bennchong, 2021-06-09:
Hi there,

I have used this plugin successfully on local instances of octoprint, however I never got to make it work with octoeverywhere hosted instances.

Issue is that checking API key never ends, and if I close the dialogue box and reopen the settings for Octoprint plugin, it will show "Checking API Key...". 

[image]

Solutions tried:
1. Using 443 port for https

Logs are showing some things I do not understand, might not be relevant

```
2021-06-09 12:59:02,023 - DEBUG - [MainThread] OctoPrintPlugin.DiscoverOctoPrintAction.testApiKey [263]: Trying to access OctoPrint instance at https://*************.octoeverywhere.com:443/ with the provided API key.
2021-06-09 13:03:17,676 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users//AppData/Roaming/cura/4.9/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/DiscoverOctoPrintAction.qml:574:25: Unable to assign [undefined] to bool
2021-06-09 13:03:17,678 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users//AppData/Roaming/cura/4.9/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/DiscoverOctoPrintAction.qml:573:25: Unable to assign [undefined] to bool
2021-06-09 13:03:17,680 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users//AppData/Roaming/cura/4.9/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/DiscoverOctoPrintAction.qml:562:25: Unable to assign [undefined] to bool
2021-06-09 13:03:17,682 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users//AppData/Roaming/cura/4.9/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/DiscoverOctoPrintAction.qml:563:25: Unable to assign [undefined] to bool
2021-06-09 13:03:17,684 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users//AppData/Roaming/cura/4.9/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/DiscoverOctoPrintAction.qml:551:25: Unable to assign [undefined] to bool
2021-06-09 13:03:17,686 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users//AppData/Roaming/cura/4.9/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/DiscoverOctoPrintAction.qml:488:29: Unable to assign [undefined] to bool
2021-06-09 13:03:17,687 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users//AppData/Roaming/cura/4.9/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/DiscoverOctoPrintAction.qml:479:25: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
2021-06-09 13:03:17,689 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users//AppData/Roaming/cura/4.9/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/DiscoverOctoPrintAction.qml:479:25: QML Connections: Cannot assign to non-existent property "onInstanceAvailablePowerPluginsChanged"
2021-06-09 13:03:17,691 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users//AppData/Roaming/cura/4.9/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/DiscoverOctoPrintAction.qml:473:29: Unable to assign [undefined] to bool
2021-06-09 13:03:17,693 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users//AppData/Roaming/cura/4.9/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/DiscoverOctoPrintAction.qml:455:25: Unable to assign [undefined] to bool
2021-06-09 13:03:17,695 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users//AppData/Roaming/cura/4.9/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/DiscoverOctoPrintAction.qml:444:25: Unable to assign [undefined] to bool
2021-06-09 13:03:17,696 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users//AppData/Roaming/cura/4.9/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/DiscoverOctoPrintAction.qml:352:21: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
2021-06-09 13:03:17,698 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users//AppData/Roaming/cura/4.9/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/DiscoverOctoPrintAction.qml:352:21: QML Connections: Cannot assign to non-existent property "onAppKeyReceived"
2021-06-09 13:03:17,700 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users//AppData/Roaming/cura/4.9/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/DiscoverOctoPrintAction.qml:314:29: Unable to assign [undefined] to bool
2021-06-09 13:03:17,702 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users//AppData/Roaming/cura/4.9/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/DiscoverOctoPrintAction.qml:191: TypeError: Cannot read property '-1' of undefined
2021-06-09 13:03:17,704 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users//AppData/Roaming/cura/4.9/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/DiscoverOctoPrintAction.qml:178: TypeError: Cannot read property 'length' of undefined
2021-06-09 13:03:17,706 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users//AppData/Roaming/cura/4.9/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/DiscoverOctoPrintAction.qml:84:17: Unable to assign [undefined] to QString
2021-06-09 13:03:17,707 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users//AppData/Roaming/cura/4.9/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/DiscoverOctoPrintAction.qml:627:25: Unable to assign [undefined] to bool
2021-06-09 13:03:17,824 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Program Files/Ultimaker Cura 4.9.0/plugins/MachineSettingsAction/MachineSettingsAction.qml:81:27: QML MachineSettingsExtruderTab: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
2021-06-09 13:03:20,088 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Program Files/Ultimaker Cura 4.9.0/plugins/MachineSettingsAction/MachineSettingsPrinterTab.qml:22:5: Unable to assign [undefined] to int
2021-06-09 13:03:20,090 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Program Files/Ultimaker Cura 4.9.0/plugins/MachineSettingsAction/MachineSettingsExtruderTab.qml:29:5: Unable to assign [undefined] to int
2021-06-09 13:03:20,111 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users//AppData/Roaming/cura/4.9/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/DiscoverOctoPrintAction.qml:479:25: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
2021-06-09 13:03:20,114 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users//AppData/Roaming/cura/4.9/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/DiscoverOctoPrintAction.qml:352:21: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
2021-06-09 13:03:20,116 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users//AppData/Roaming/cura/4.9/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/DiscoverOctoPrintAction.qml:338:21: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
2021-06-09 13:03:20,141 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users//AppData/Roaming/cura/4.9/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/DiscoverOctoPrintAction.qml:620: TypeError: Cannot read property 'getId' of undefined
2021-06-09 13:03:20,689 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Users//AppData/Roaming/cura/4.9/plugins/OctoPrintPlugin/OctoPrintPlugin/qml/DiscoverOctoPrintAction.qml:380: TypeError: Cannot read property 'getId' of undefined
2021-06-09 13:06:51,707 - DEBUG - [MainThread] OctoPrintPlugin.DiscoverOctoPrintAction.testApiKey [263]: Trying to access OctoPrint instance at https://*************..octoeverywhere.com:443/ with the provided API key. 
```

fieldOfView, 2021-06-22:
Could you post a link to a full Cura.log? It is just starting to get interesting with the last line of the snippet you posted.

bennchong, 2021-06-26:
This is the full log excluding non-octoprintplugin related logs. There wasn't any more info after the last line. Unless there is a setting to show more logs within cura?

fieldOfView, 2021-06-27:
The full log may show me more than what you may think I want to see.

bennchong, 2021-06-27:
[cura.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/6721736/cura.log)

I've uploaded the log here, thanks

fieldOfView, 2021-06-27:
Thanks, I'll have a look soon.

fieldOfView, 2021-06-27:
I believe you have to set up a "shared connection" in OctoEverywhere:
https://octoeverywhere.com/sharedconnections

fieldOfView, 2021-06-27:
What happens under the hood is that OctoEverywhere says that Cura is not "logged in" to OctoEverywhere.com when it is trying to access your octoprint instance. I cannot find documentation how Cura could "log in" to OctoEverywhere,

bennchong, 2021-06-29:
I see. I thought an locally hosted octoprint end points would behave similarly to an octoeverywhere instance portal, but probably thats a too simplistic view. New to this place, thanks for upgrading this issue to an enhancement for future updates.

fieldOfView, 2021-07-08:
> I thought an locally hosted octoprint end points would behave similarly to an octoeverywhere instance portal,

So did I, but apparently a super-secret handshake is required.

QuinnDamerell, 2022-02-23:
Hey! I wanted to quickly follow up on this, I'm the developer behind OctoEverywhere.

You can use OctoEverywhere with this plugin and Cura right now! You're on the right path, all you need to do is create a shared connection and add it to the plugin as the hostname. Using a shared connection you have a unique subdomain that allows anyone who has access to it access to the tunnel to your OctoPrint. However, since OctoPrint has it's own login system, you still need to give the Cura plugin access to OctoPrint the same way you would locally, but using an OctoPrint Access Key.

The more modern way of setting up the connection would require plugin support, but OctoEverywhere has a feature called App Connections that allows 3rd party OctoPrint apps super-easy integration. From a user perspective, they would click a button in the plugin, log into OctoEverywhere in the pop-up window, and they are done! The OctoEverywhere App Connection portal will return all of the required and secure connection information to the Cura plugin, including the OctoPrint API key!

I would love to enable you to add the integration @fieldOfView by doing anything you need. I have full API docs online you can follow, it should be very easy. 
https://octoeverywhere.stoplight.io/docs/octoeverywhere-api-docs/ZG9jOjcyNTE0ODU-app-connection-overview

Let me know if I can help in any way!