fieldOfView/Cura-OctoPrintPlugin #239: Option to skip wait on gcode analysis

H4R0, 2021-06-16:
Could we get a option in settings to disable wait for gcode analysis ?

Gcode analysis takes some time but runs fine parallel to the start sequence (printer heating, nozzle cleanup, skirt etc.)

And even if it still runs during print there is no noticeable slow down. 

Print time becomes available after it finished analyzing in Cura automatically.

I don't see a reason to wait for it, I always skip it and there are no issues.

https://github.com/fieldOfView/Cura-OctoPrintPlugin/blob/7bd73946fbf22d18337dc900a81a011ece26bee0/OctoPrintOutputDevice.py#L1161

Add skip option here https://github.com/fieldOfView/Cura-OctoPrintPlugin/blob/7bd73946fbf22d18337dc900a81a011ece26bee0/OctoPrintOutputDevice.py#L1154

reference:
https://github.com/fieldOfView/Cura-OctoPrintPlugin/commit/e07208fa2c7343afca7db40418283c98f4cde208

https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/97

fieldOfView, 2021-07-06:
The next version of the plugin will look at how OctoPrint is configured. See `Settings -> Features`:

[image]

If this setting is set to Always or Never, the plugin will no longer wait for analysis. If it is set to "on idle", it makes no sense to always skip waiting for the analysis to complete; doing so stops the analysis, so you might as well disable the estimation plugin altogether.

fieldOfView, 2021-11-11:
I am finishing up the next version of the plugin, which implements a check to see if OctoPrint allows gcode analysis while printing (see above).

Please help me test this feature by downloading this [development snapshot](https://storage.googleapis.com/um-cloud-production-cura-packages/v1/packages/OctoPrintPlugin/FF_CBKkntJes5ApbMoHWqgklxPoLeubdesoMFX62xHHo/OctoPrintPlugin-v7.0.0-2021-11-09T17%3A14%3A47Z.curapackage?X-Goog-Algorithm=GOOG4-RSA-SHA256&X-Goog-Credential=cura-packages-api%40um-cloud-production.iam.gserviceaccount.com%2F20211109%2Fauto%2Fstorage%2Fgoog4_request&X-Goog-Date=20211109T171503Z&X-Goog-Expires=604800&X-Goog-SignedHeaders=host&X-Goog-Signature=68c6272055900f6c5a7db3a029fc54ba8adb7e90b0a3bd0ac52b071b036aa76b04af2b5a0aab494f6c1c93aadf13c3fcb8bce93635439121b1ad52d64c053cff767bf336edec6e2cb40a3f10b72d871f67dc983f351e842ae7f7da6d08d93cf9aa99ffc2f15abc9f5414dc295b08c42d3973c924fd4ddc8f1b05ead5562ee471effa9890234e09003eb244fbfe47fde9a295fdb21eae4bfcb7fc59169761747e8f4a102fde5232c91f0f0242fa43a1de2f5944c9045bb95fa4469cb3fcb3c41c2ed5e27f74d686c5ca2da373e95ddedfb146f6fbb874b7c9f4fc8617fabd2d3ad6b91f9be68183d7e712f781aac7dbe17e00df479378392513976c2d5b412548) and dropping it on the Cura buildplate to install it.

H4R0, 2021-11-15:
I haven't printed a lot in the last couple of weeks.

I will see if I can test it by tomorrow evening.

closed 2021-12-21 by fieldOfView