fieldOfView/Cura-OctoPrintPlugin #240: no connection to Octoptint due to false port

mic62, 2021-06-21:
I can't access my printer hanging on a Raspi from Cura via Octoprint plugin due to the fact, that I have multiple printers connected to my Pi. Theprinters doe all have the same IP (obviously), but diferent ports.
The plugin automatically chooses port 80 and doesn't let me change it to the correct one.

fieldOfView, 2021-06-22:
The plugin uses whatever port is advertised by bonjour. You can either turn off bonjour discovery in the plugin and "Add" them manually (which allows setting the port), or you can configure OctoPrint to advertise the proper port for each instance. To do that, you need to edit the config.yaml for each instance with the correct `plugins/discovery/publicPort`.

```
plugins:
  discovery:
    publicPort: 80
```

mic62, 2021-06-22:
Found it! Thanks a lot for the speedy response!

closed 2021-06-23 by fieldOfView