fieldOfView/Cura-OctoPrintPlugin #243: Cura 4.10 - Printer does not accept commands

tuplink, 2021-07-20:
Versions
  OctoPrint: 1.6.1
  Printer:      Ender 5
  Plugin:       3.5 / Master
  Cura          4.10
  
Webcam Feed loads but Monitor says **Printer does not accept commands**
stderr.log  Shows no errors
this setup worked prior to upgrading to Cura 4.10 from 4.8

https://octoprint/api/printer?apikey=XXXXXXXXXXXX
  `{"sd":{"ready":false},"state":{"error":"","flags":{"cancelling":false,"closedOrError":false,"error":false,"finishing":false,"operational":true,"paused":false,"pausing":false,"printing":true,"ready":false,"resuming":false,"sdReady":false},"text":"Printing"},"temperature":{"bed":{"actual":55.0,"offset":0,"target":55.0},"tool0":{"actual":195.0,"offset":0,"target":195.0}}}`

tuplink, 2021-07-20:
STDERR.log and cura.log
https://pastebin.com/NDgQZHvd

fieldOfView, 2021-07-20:
I think this is a duplicate of the elusive (to me anyway) #136.

fieldOfView, 2021-07-20:
> this setup worked prior to upgrading to Cura 4.10 from 4.8

Have you tried reinstalling 4.8? It would be interesting to know if this is indeed limited to Cura 4.10

tuplink, 2021-07-20:
so i went back to 4.8 with the appdata/cura/* settings that existed befor and all still worked.     InstanceID hostname port 80 and path.     copied the settings from 4.8 to 4.10, still dose not accept commands.     changed the reverse proxy settings to use ssl and change port to 443.     this now works.   im guessing something happened that make HTTP 301 responses invalid, as i have a reverse proxy setup for octoprint.     probably not an issue with OctoPrintPlugin, and more so a change in Cura that is causing this. however documenting this here incase anyone else has the same issue. 

Just as a clarification the exact setting in OctoPrintPlugin did not work with 4.10    had to Show reverse proxy options and select Use HTTPS then change the port to 443.

Issue Resolved

fieldOfView, 2021-07-20:
Ah, in that case, see https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues/168

Between 4.8 and 4.10, Cura updated the Qt version it uses. The plugin uses the networking functionality that is offered by Qt, so if something changed that is likely the cause.

I'll close this issue as a duplicate of #168, and reopen that one. I'll see if there is anything I can do about this.

closed 2021-07-20 by fieldOfView