fieldOfView/Cura-OctoPrintPlugin #244: Add material used to Gcode for Filament Manager Plugins in OctoPrint

NilsRo, 2021-08-03:
I like to use a Filament Manager Plugin in OctoPrint to get warned if a spool contains not enough filament to finish the print. The actual filament manager could not automatically detect the material used so it has to be selected in the Webfrontend manually. As I am using materials in Cura it would be very helpful if Cura can add the filament used in the GCode file. But according this list http://files.fieldofview.com/cura/Replacement_Patterns.html the material name is not included. Material GUID cannot be used as it is the same for every custom material. This is also discussed many times for Cura itself but I did not find any discussion to add this for feeding a Filament Manager in OctoPrint. So it goes from a more or less statistical information to a useful information to manage your filament.  

GCode has to store the filament used (included important information like color,...). I will also add a feature request for Cura itself, but it is more useful combined with OctoPrint Filament Manager.

fieldOfView, 2021-08-03:
>  Material GUID cannot be used as it is the same for every custom material.

The material GUID is only the same if you keep duplicating materials to create a new one. If you üse the "Create" button instead of "Duplicate", you get a new GUID for each newly created material. For materials that you "Duplicate"d before, there's an "Unlink" button somewhat hidden below the material properties:
[image]

fieldOfView, 2021-08-03:
If the material GUID is unique per material (which it should be), would that suffice? I don't mind adding functionality to work with specific plugins in OctoPrint, but for maintainability I would like to keep that to a minimum.

NilsRo, 2021-08-05:
I like to get the properties linked as for example I created 4 new materials out of PETG like to adapt some properties only (same GUID). Or Made a brand PETG material and like to duplicate it for every color I own. What you describe is a workaround for the case I have in mind to identify the different filament roles by e.g. color which is important for the filament manager. 

So If I duplicate green PETG, yellow PETG, ... from generic PETG I like to change the print temperature for generic PETG only. Because of that it would e helpful to get a human and computer readable comment in the Gcode instead. This could be also used to add it to OctoPrint metadata to show the filament the file is made for (as a new additional idea).

Helpful for: 

- manually check for which material the GCode was created
- show material in OctoPrint Filebrowser (more comfort)
- make material identifiable for a filament manager to monitor the filament left on a role

fieldOfView, 2021-09-24:
The Cura PR is merged, so you will be able to add the new replacement patterns from Cura 4.12 onwards.

closed 2021-10-04 by fieldOfView