fieldOfView/Cura-OctoPrintPlugin #247: Plugin nstalled but not showing in Cura 4.10 and 4.11

InnoProd, 2021-09-08:
Was delighted to find this plugin. We may switch our Lulzbots to Cura 4.x if I can get the Octos to connect. 

Marketplace shows plugin installed: 
[image]

But after multiple restarts, the Connect Octoprint button is not showing: 
[image]

Perhaps I'm missing something obvious, or I need to wait for the plugin to catch up with 4.10 and 4.11

Thank you for your work!

fieldOfView, 2021-09-09:
What printer do you have?

I see a `Connect via Network` button. This only shows up if you selected an Ultimaker 3, S5 or S3 from the list of printers when you added your printer to Cura. Those printers are not compatible with OctoPrint. You will also get all sorts of other issues, such as saving .ufp instead of .gcode, and printing with ultimaker-specific settings which may not be optimal for your printer.

InnoProd, 2021-09-09:
> What printer do you have?
>
> I see a `Connect via Network` button. This only shows up if you selected an Ultimaker 3, S5 or S3 from the list of printers when you added your printer to Cura. Those [...]

Ah, did not understand the workflow for adding printers to Octoprint. I assumed the Connect Octo button would be present regardless of the printer profile. The button shows once I add a non-ultimaker printer and it's picking up my Octo instances just fine. Pilot error, I apologize for the unnecessary issue.  Thank you for the clarification!

closed 2021-09-09 by InnoProd