fieldOfView/Cura-OctoPrintPlugin #248: Cura OctoPrint Plugin:  IP address of stored OctoPrint instances defaults back to 172.17.0.1 ("cura octoprint not available")

wnsander232, 2021-09-19:
Issue:  IP changed on stored OctoPrint instances in Cura

Workaround:  Delete and re-add all OctoPrint instances in Cura.

Detail:  I made the mistake of updating my Linux server and Cura at the same time (I know better).  After updates, Cura could not find any of the OctoPrint servers warning "cura octoprint is not available".  However, I could get to  OctoPrint through the web interfaces and even print by dragging the file onto the web interface and selecting print.  After losing a few strands of hair I realized all of my instances, 10 total, had different IP's than originally configured.  I suspect this is a Cura issue but wanted to let the community know.  I deleted then re-built all the instances.  Everything is working now.

wnsander232, 2021-09-19:
Issue:  No printer control from within Cura.

Workaround:  Use OctoPrint for control.  No workaround from within Cura

Detail:  There is a secondary issue.  Even when connected to Cura via Settings the control feature in the Monitor tab (Cura) maintains the 172.17.0.1 IP address, which prevents connectivity to the OctoPrint instance.

fieldOfView, 2021-09-21:
How did you add the OctoPrint instance to Cura? Using the "Add" button, or using automatic discovery? If you used the "Add" button, and you entered the (now invalid) IP address, then it is expected that Cura can no longer find the printer. You would have to use the "Edit" button. 

If you used the automatic discovery, and Cura is still using the old IP address, then OctoPrint is telling Cura to use the old IP address, and the OctoPrint Connection plugin can't do much about that. Zeroconf records are not cached in any way between sessions in Cura.

wnsander232, 2021-09-21:
I understand what you are implying, and that would definitely cause an issue such as this.  However, in my case the 172 IP range would never have been used on my internal network and I do not have remote access enabled.  Its also worth noting my server has been running the same 192.xxx.xxx.xxx IP range since it was constructed 3-years ago.  A bit more detail:

1.  I manually configured instances of OctoPrint within cura (Settings/Printer/Manage Printer/Connect Octoprint/"Add"/"Add printer by IP).  
2. Everything has been working perfectly for at least a year (maybe two... hard to tell in COVID limbo)
3. I ran Linux updates on my server, which is running the ten virtual instances
4. I updated Cura to 4.11.0
5. OctoPrint integration quit working due to my manually configured IP addresses defaulting to the 172 IP range.  
6. I deleted ALL instances of Octoprint, ensuring "Automatically discover local OctoPrint instances" was unchecked.
7. I reconfigured OctoPrint instances using the "Add" button.
8. Send to OctoPrint functions began working again.  However, I still had issues with the 172.x.x.x ip addresses within the Monitor pane.
9. After about a day, this last issue seems to have magically worked itself out.  No reboots... no closing of apps... makes no sense, but it is what it is.  

I have updated Cura and Linux multiple times in the last three years and have never had to reconfigure my OctoPrint instances within Cura.  As with anything network/computer related, this may just be a fluke without logs to track down root cause.  Everything seems to be working now.  Hopefully this thread helps anyone who experiences the same issue.

Thank you for such a great app!  

-N

fieldOfView, 2021-09-21:
Did you press the "Connect" button between step 7 and 8?

fieldOfView, 2021-10-04:
@wnsander232, did my question help you find a solution?

orangeworx, 2021-10-04:
I'm having a very similar issue but this has happened with every Cura update.
I start from scratch everytime, it's actually quite annoying. But hey, convenience while it lasts.

The difference for me aside from the wonky ip range, is that there's always a second printer that loses the API "Request" button.
and unless I remove to "connectable" instance, I just can't get a connection.
Auto-search also just stopped working after last update.

I know this isn't "scientific". I haven't spent more than a couple hours trying things.
But here's what I remember doing...
After updating and running into no printers, removing cura and everything that it leaves behind. Reinstalled, updates and octoprint plugin.
Added printers locally and attempted connections. The request button disappearing, well I'm not sure if that's the real case, it might be just pushed out of view, the key input field spans whole window. No API key works, tried main key as well as app keys that I've generated. I know that worked fine this way for the other printer.
open in browser, does open the instance's page.

My apologies for my crappy reporting... I shouldn't have been messing with that with my state of affairs but here I am.

fieldOfView, 2021-11-09:
I am closing this issue due to lack of response from @wnsander232 

@orangeworx if you are still having an issue connecting, please open a new issue and try to be as "scientific" as possible so I can help you figure out what is going on.

closed 2021-11-09 by fieldOfView