fieldOfView/Cura-OctoPrintPlugin #253: Feature : Display the content of the octoprint filesystem

Turiok, 2021-11-17:
When We can add a path to upload the file on Octoprint. It'll be great to see the content of the actual filesystem which is stocked the existing gcode.
I think, the exploration will be only in read only. To be sure to not delete existing file. But only with the purpose to find the right path for the file we want upload.

Combinacijus, 2022-04-02:
I wanted to ask for a feature to remember previous `Path` text between prints and between Cura restarts but OP's feature request is better and my feature request would be obsolete
[image]

fieldOfView, 2022-04-03:
In the mean time, can I recommend the Print Job Naming plugin, which will let you set a default path (either with patterns or without)?

Combinacijus, 2022-04-03:
@fieldOfView Thanks that solves default path problem plus introduces some useful features
And for others tutorial how to setup **Print Job Naming** plugin:

Download the plugin and restart Cura: <https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/CustomJobPrefix?action=addToCura>

Setup plugin:

[image]
[image]

And now path is filled automatically and filename modified as specified:

[image]

fieldOfView, 2022-04-03:
If you want to save one or two clicks, there should also be a small folder icon next to the job name in the lower left corner of the viewport.