fieldOfView/Cura-OctoPrintPlugin #254: Integrate with FillamentManager

impatman, 2021-11-17:
Source:  https://plugins.octoprint.org/plugins/filamentmanager/

I'd like to be able to select which filament I'm about to print without having to go into octoprint first. I'm often changing the filament between prints and this would save me a few steps. 

I'd imagine it'd be a popup and/or a dropdown to select before printing or in the monitor tab.

cruzer619, 2022-01-17:
Yes please.. will like this enhancement as well

fieldOfView, 2022-01-18:
Cura already has material selection. What is needed beyond what Cura already does?

cruzer619, 2022-01-19:
"to my knowledge" materials selection doesn't track what what filament is being used.  also doesn't stop you from printing if there isn't enough filament to print the job (from what it stores in its database).  Also in my situation since I have 3 printers.  They share the same database.  So keeps track of filament used no matter what printer.  And I don't slice on the same computer.  Sometimes on my main computer, sometimes computer at work.  Then I  use telegram plugin to print remotely.  So can't really use one instance of Cura to hold my filament data.  Octoprint is running on all 3 printers.  So this plugin really suites my needs and so does yours.  As @impatman was stating.  if I changed filaments, I would have to open up octoprint on web browser first.  Select the filament I want to use.  Then go back to cura and then press "print with octoprint" to print my object.  I hope that all makes sense.

kylemallory, 2024-01-05:
Another upvote for this.  FilamentManager is a great tool.  I only have a single printer; my sons and I all use the same printer but submit jobs from each of our own PCs.  FilamentManager is a great tool for keeping track of the remaining filaments, that is tied to the printer and not the Cura installation.  This doesn't need to be complex, just a simple dropdown of available filaments.  Ideally, it would auto-select the current filament.  As is, we slice in Cura, switch to Octoprint, select/confirm the filament, and then go back to Cura to actually submit the job.