fieldOfView/Cura-OctoPrintPlugin #255: "Connect OctoPrint" only for first printer in preferences available

echterhoff, 2021-11-22:
Hi there!

Thanks for your plugin and your effort. As I have running two octopi's since yeasterday, I'd like to control both printer through cura. But I could not find the button in the preferences in context to the desired printer profile.
While switching between the active instances, I discovered, the button only shows up on the first printer profile.
I managed to manually setup the configuration in the .global.cfg-files to get both printer hooked up to their octoprint instances through your plugin.

Buuuuuut it would be rather nice to have this sorted. Sadly I am not familiar with python but I will also take a look at what might have gone wrong. (Two hours later. - Nope, there is no way I can figure that out. It must be something withing DiscoverOctoPrintAction.qml and class DiscoverOctoPrintAction...)

Thanks in advance.

Lars

fieldOfView, 2021-11-23:
The "Connect to OctoPrint" button, like other buttons such as the "Machine Settings" button, is only available for the currently "active" printer. In the Printers pane of the preferences, you can press the "Activate" button to activate a printer.

This is how Cura works, and not something this plugin can change.

closed 2021-11-23 by fieldOfView

echterhoff, 2021-11-23:
Okay. The problem is/was within the machine definition. It has nothing to do with position. I am aware of the need to active the printer to access the button.
The Qidi X-Max definition file is missing the supports_usb_connection metadata, since it lacks support of usb as an interface.

    def _onContainerAdded(self, container: "ContainerInterface") -> None:
        # Add this action as a supported action to all machine definitions
        if (
            isinstance(container, DefinitionContainer) and
            container.getMetaDataEntry("type") == "machine" and
            container.getMetaDataEntry("supports_usb_connection")
        ):

            self._application.getMachineActionManager().addSupportedAction(container.getId(), self.getKey())

This simply prevented the buttom from showing up. Sorry for bothering you with my problems.

fieldOfView, 2021-11-23:
Yes, that would also do it. The logic is that printers with no usb port can not interface with OctoPrint in the first place.