fieldOfView/Cura-OctoPrintPlugin #260: Cura 4.12 (also 4.13 Beta) no longer loading/showing OctoPrint plugin

tophee, 2021-12-22:
The OctoPrint plugin is no longer working for me. I was on Cura 4.12 and then upgraded to 4.13 beta but the problem persists. I have plugin version 3.6.0 installed via the marketplace and it is not offering to print to print via octoprint like it used to do. Since the plugin is not even in the Cura "Extensions" menu, I guess it simply fails to load. 

I'm on MacOS 12.0.1.

I have tried figuring this out for hours now but I'm not a programmer so I'm not getting anywhere. I have uninstalled the blugin many times and deactivated various other plugins, but to no avail. 

I include the [cura.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/7765232/cura.log) file which seems to include som hints but I don't understand enough to make any sense of it.

fieldOfView, 2021-12-23:
> Since the plugin is not even in the Cura "Extensions" menu, I guess it simply fails to load.

The plugin has never had an entry in the Extensions menu. Not all plugins do.

Thanks for including the cura.log file. From the looks of it, the plugin is loading just fine, though there seems to be an issue with the "bonjour"/"zeroconf" discovery of the octoprint instance. 

On the "Printers" pane of the Preferences, there should be a "Connect to OctoPrint" button. When you press that button, the "Connect to OctoPrint" dialog should pop up. Is your printer listed there? How about when you press the "refresh" button?

tophee, 2021-12-23:
Thanks for looking into this. I spent some time troublshooting this and ended up reflashing my octopi to the latest version (0.18) because that supports python3. Much was pointing into the direction of python being the issue, but I'm not sure if that actually was the root cause. I actually had to flash to a new SD-card because the old one didn't pass the read-test after flashing, so chances are that it was simply the SD-card causing errors here and there. 

I've been going down so many rabbit holes during the day that I don't exactly remember what I did in what order but since all I did was on octopi and not on cura or the plugin, it is safe to assume that the issue was not caused by your plugin.

I was confused, though, to find the plugin settings in the printer profile (although I obviously configured them there a year or two ago). As I think about it, it makes sense to have them there, but at least in my case, it would have helped me if there had been a hint about where to configure the plugin. Perhaps in the description of the plugin on the cura marketplace? Or, ideally, in the extensions menu. It could just link to the printer settings or something.

closed 2021-12-23 by tophee