fieldOfView/Cura-OctoPrintPlugin #261: Plugin not starting using Python 3.10

duckunix, 2021-12-28:
OS: ArchLinux fully up to date
Python: 3.10.1
Cura Version: 4.11.0 [cura.log](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/7786344/cura.log)
Plugin version: 3.6.0

Installed the plugin via the Market Place, and restarted Cura.  I did not see the option to connect to my OctoPrint, so I went back into the Market Place, and saw the plugin installed, but disabled.  I have attached a copy of the log on startup, and there are some python errors.  My Python-fu is weak, but given some guidance I should be able to help fix.

Thanks!

fieldOfView, 2021-12-28:
> Installed the plugin via the Market Place

Are you sure you installed it via the MarketPlace? It does not seem to be installed to the place that the MarketPlace would normally install plugins.

I think the problem is I forgot to update the CMakeList.txt file, which is not used by MarketPlace installs, but it is used by some package managers.

duckunix, 2021-12-28:
Ack.  You are right.  I had installed a package from the AUR, and it seems to not work there.  Removed that package, and it is all good now.  

Thanks!

fieldOfView, 2021-12-29:
Thanks for letting me know.

I am reopening (and renaming) your issue until I have fixed the CMakelist.txt file so the package in AUR works again.

duckunix, 2021-12-29:
Thanks for trying to fix the AUR package!

commit bc7ae9c8 (2021-12-29): Add missing files to CMakeList.txt

Fixes #261
closed 2022-05-03 by fieldOfView