fieldOfView/Cura-OctoPrintPlugin #262: The server encountered an internal error

ajs256, 2021-12-31:
When sending a sliced file to OctoPrint, after uploading successfully, a message appears about an "internal error".

Plugin version: latest as of https://github.com/fieldOfView/Cura-OctoPrintPlugin/commit/105f540d39517e8bb25d8cd999849e995662653e
Cura 4.12.1

Octoprint system info: [octoprint-systeminfo-20211231211141.zip](https://github.com/fieldOfView/Cura-OctoPrintPlugin/files/7797049/octoprint-systeminfo-20211231211141.zip)

Relevant lines of Cura log:
```
2021-12-31 13:04:47,338 - ERROR - [MainThread] Cura-OctoPrintPlugin.OctoPrintOutputDevice._onUploadFinished [1544]: OctoPrintOutputDevice got an 500 error uploading to http://192.168.163.155:80/api/files/local
2021-12-31 13:04:47,339 - ERROR - [MainThread] Cura-OctoPrintPlugin.OctoPrintOutputDevice._onUploadFinished [1550]: {"error":"The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application."}
```

Relevant lines of OctoPrint log:
```
2021-12-31 21:04:47,213 - octoprint.filemanager.analysis - INFO - Starting analysis of local:OPI3MMK3_joycon_strap_holder.gcode
2021-12-31 21:04:47,215 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python3 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 --bed-z=0.0 /home/pi/.octoprint/uploads/OPI3MMK3_joycon_strap_holder.gcode
2021-12-31 21:04:47,285 - octoprint.printer.standard.job - INFO - Print job selected - origin: local, path: OPI3MMK3_joycon_strap_holder.gcode, owner: ajs256, user: ajs256
2021-12-31 21:04:47,307 - octoprint - ERROR - Exception on /api/files/local [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/util/flask.py", line 1552, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/flask_principal.py", line 199, in _decorated
    rv = f(*args, **kw)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/files.py", line 666, in uploadGcodeFile
    target,
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/files.py", line 623, in fileProcessingFinished
    selectAndOrPrint(filename, absFilename, destination)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/files.py", line 642, in selectAndOrPrint
    user,
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/printer/standard.py", line 727, in select_file
    tags=kwargs.get("tags", set()) | {"trigger:printer.select_file"},
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 1615, in selectFile
    filename, self._currentFile.getFilesize(), False, user=user
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/printer/standard.py", line 1615, in on_comm_file_selected
    self._create_estimator()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/printer/standard.py", line 202, in _create_estimator
    self._estimator = self._estimator_factory(job_type)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrusaETAOverride/__init__.py", line 57, in __init__
    super(pETAPrintTimeEstimator, self).__init__(job_type)
TypeError: super(type, obj): obj must be an instance or subtype of type
2021-12-31 21:05:16,332 - octoprint.filemanager.analysis - INFO - Analysis of entry local:OPI3MMK3_joycon_strap_holder.gcode finished, needed 29.12s
2021-12-31 21:11:53,047 - octoprint.server.heartbeat - INFO - Server heartbeat <3
```

ajs256, 2021-12-31:
Never mind, just had to restart the OctoPrint server

closed 2021-12-31 by ajs256

fieldOfView, 2022-01-01:
This seems to have been an issue in the PrusaETAOverride plugin in OctoPrint. Please let me know if it resurfaces.